AFM-SPM/TopoStats #10: Add installation instructions to readme

bobturneruk, 2020-11-02:
On Windows, these might form the basis, but need to be checked.

# Install

* Install 32 bit Anaconda (https://www.anaconda.com/distribution/#windows) - if you have 64 bit installed as well, make sure you’re using the 32 bit version
* Install 32 bit Gwyddion for Windows (http://gwyddion.net/download.php#stable-windows)
* Download the contents of this Google Drive folder which contains additional prerequisites (https://drive.google.com/drive/folders/1A3Y4M9kzbB7Hyj5J2mbV7jr1aTaLeIna?usp=sharing)
* Use Anaconda Navigator to import an environment from gwyconda.yml (in the Google Drive)
Identify the path to the copy of Python in the new conda environment. A way to do this:
    * Open Anaconda Powershell prompt
    * Type `conda activate gwyconda` (this puts you in the correct environment)
    * Type `Get-Command python`
* Install (for all users) from the Google Drive Folder making sure to select the folder containing the version of python identified in the previous step
    * pycairo-1.8.10.win32-py2.7.msi
    * pygobject-2.28.3.win32-py2.7.msi
    * Pygtk-2.24.0.win32-py2.7.msi

# Test

* Open Anaconda Powershell prompt
* Type `conda activate gwyconda` (this puts you in the correct environment)
* Type `cd 'C:\Program Files (x86)\Gwyddion\bin'` or `cd <your_gwyddion_path>\bin`
* Type `python` (you should see something like:
```
Python 2.7.16 |Anaconda, Inc.| (default, Mar 14 2019, 15:49:40) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
```
* Type `import gwy`

If you get no errors, you should be good to go…

bobturneruk, 2020-11-23:
Also add instructions to clone the repository.

derollins, 2020-11-30:
> Also add instructions to clone the repository.

Here how to clone the repository in command line, haven't worked out how to do it in PyCharm yet

# Clone the Repository 

- Ensure that you have git installed (download git here: https://git-scm.com/downloads)
- In command line or powershell navigate to the folder where you want to install Topostats,
- Type `git clone https://github.com/afmstats/TopoStats.git`. This will make a copy of the code in a folder called TopoStats.
- Make a branch to work on by typing `git checkout -b <new branch name>`

bobturneruk, 2020-12-01:
Good work @derollins! In pycharm you need to close your current project then start here:
[image]

MaricruzFernandez, 2021-01-27:
Dear @bobturneruk 
Can you share me the Google Drive folder? 
(https://drive.google.com/drive/folders/1A3Y4M9kzbB7Hyj5J2mbV7jr1aTaLeIna?usp=sharing)
Can you give me access?
I'm trying to install all on Windows. 
Many thanks
E-mail.: fernandezm@exa.unrc.edu.ar

bobturneruk, 2021-01-27:
Hi @MaricruzFernandez please use this source instead https://sourceforge.net/projects/gwyddion/files/pygtk-win32/

I will edit the first comment.

bobturneruk, 2021-01-27:
I've just retried building with conda on windows 10 and I'm having major problems with pycairo.

MaricruzFernandez, 2021-01-28:
Dear

Unfortunately, I am not being able to install too much. I am not very familiar with programming. Please, if at any time you have a step by step on how to do it, I would appreciate it if you can send it.
Many thanks

Maricruz

bobturneruk, 2021-01-28:
I think the instructions here are probably no longer working. I'm going to have a think about what the best approach is. @derollins and @Jean-Du have installed recently and may have some useful advice?

derollins, 2021-01-28:
Hi @MaricruzFernandez ,

This process has been working for me. What part are you stuck on?

If it helps here are direct links to the relevant 32 bit versions of Anaconda and Gwyddion.

https://repo.anaconda.com/archive/Anaconda2-2019.07-Windows-x86.exe
http://gwyddion.net/download.php#stable-windows

You need to make sure you are using Python 2 and you also need to ensure you install the dependencies for all users.

Hope this helps, let us know if you need any more help.

MaricruzFernandez, 2021-01-28:
I already installed 32 bit versions of Anaconda and Gwyddion. Also, I use Python 2 and I installed : pycairo-1.8.10.win32-py2.7.msi
pygobject-2.28.3.win32-py2.7.msi , Pygtk-2.24.0.win32-py2.7.msi (this last three I downloaded from internet).
Now, I'm doing tje steps that are mentioned above but I got error:

(base) PS C:\> conda activate gwyconda
(gwyconda) PS C:\> Get-Command python

CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Application     python.exe                                         0.0.0.0    C:\Users\fernandezma\OneDrive\envs\gwy...

(gwyconda) PS C:\> conda activate gwyconda
(gwyconda) PS C:\> "C:\Program Files (x86)\Gwyddion\bin\gwyddion.exe"
C:\Program Files (x86)\Gwyddion\bin\gwyddion.exe
(gwyconda) PS C:\> python
Python 2.7.16 |Anaconda, Inc.| (default, Mar 14 2019, 15:49:40) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import gwy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named gwy

What could be the problem?

derollins, 2021-01-29:
After you enter `conda activate gwyconda` the second time the next command should be to change directory to the location of gwy.pyd i.e. `cd C:\Program Files (x86)\Gwyddion\bin`

If gwy.pyd is in there and the rest is set up right it should work.

bobturneruk, 2022-02-19:
I'm getting this when I try and make the conda env, which means pycairo 1.8.10 is no longer on PyPi:

```
ERROR: No matching distribution found for pycairo==1.8.10 (from -r C:\Users\vagrant\Documents\TopoStats-master\condaenv.3zxp525c.requirements.txt (line 1))
```

bobturneruk, 2022-02-19:
And I can only create the conda env from the command prompt, not powershell (see https://github.com/conda/conda/issues/5448).

bobturneruk, 2022-03-04:
Not sure if these still work, so probably don't add to readme.

bobturneruk, 2022-03-31:
Will need to be continuously updated as we move away from Gwyddion + py3.

bobturneruk, 2022-04-06:
This essentially depends on removal of Gwyddion as a dependency, at which point installation instructions should boil down to `pip install topostats`, or `pip install -e .` for development.

bobturneruk, 2022-09-13:
Installation instructions are now in readthedocs.

closed 2022-09-13 by bobturneruk