AFM-SPM/TopoStats #19: Add input (and output?) config files

bobturneruk, 2020-11-30:
YAML, TOML and JSON are potential options for this. TOML is probably the most readable, YAML is more broadly used and JSON is very widely used but has a some extra syntax which might make it harder to read. The first step down this road would be to capture what inputs are required to make topostats run. Maybe start listing these below?

derollins, 2020-12-01:
Since the stats output files are JSON is there an advantage to using them as config files or would it be better to use a different type?

bobturneruk, 2022-02-17:
I think this was resolved by #49 - if this is not the case, @SylviaWhittle please open a new issue on what still need to be done.

closed 2022-02-17 by bobturneruk