AFM-SPM/TopoStats #37: Identify "hard-coded" variables that can be input as command line arguments

JoeBeton, 2021-02-10:
There are multiple hard coded variables throughout the code that are important for optimising processing of different samples or need to changed each time the code is executed (e.g. path to input files).

These need to be identified and listed to facilitate writing a command line interface.

derollins, 2021-08-02:
Hard coded variables in addition to path:
Colour gradient 'palette'
xy, and z scales for each output image
Maximum and minimum height

JoeBeton, 2022-02-15:
Can this issue also be closed (related to #18) as it looks to me (from a fairly quick look admittedly) that all the hard coded variables are now cleaned up and a default behaviour is fairly well defined 
@Jean-Du @alicepyne @bobturneruk

bobturneruk, 2022-02-15:
May relate to #45

bobturneruk, 2022-02-17:
@SylviaWhittle may have an opinion on this.

SylviaWhittle, 2022-02-23:
I need to ensure that #45 is sorted out before this can be closed I think. I have just came across a problem with it today. Will look into it.

bobturneruk, 2022-04-06:
No current plans to resolve #45 directly on `master`, but will be fixed on `dev`. This is related https://github.com/AFM-SPM/TopoStats/issues/87

Jean-Du, 2022-04-12:
> There are also hard-coded variables at lines: 217, 254 and 280 of pygywtracing.py in the mask_outliers2 function ([read more here](http://gwyddion.net/documentation/head/pygwy/gwy.DataField-class.html#mask_outliers2)).
> 
> The hardcoded '5' relates to the lower masking threshold but would be removed when we become Gwyddion independent, but I thought I'd make people aware.
> 
> @Jean-Du Knows about this

This function is used to masks features of interest by selecting grains that deviate from the mean height by a given threshold.
Previously, the Gwyddion function "mask_outliers" was used for this purpose, but this sets the same value for the upper threshold and the lower threshold, which means if the user wants to select features that are 1σ above the mean, anything that is 1σ below the mean will also be masked. In order to deal with this, I changed it to "mask_outliers2", which allows different values for upper and lower thresholds. Since in the majority of the time TopoStats is used to select features higher than the background, I set the lower threshold to a random high number (5 in this case) for a quick fix.
In the future, if we are moving away from Gwyddion, it would be helpful to develop functions that only mask things at one end of the height distribution (eg only mask features higher than 1σ from the mean, without masking anything lower than the mean), as mask_outliers and mask_outliers 2 don't allow that at the moment.

SylviaWhittle, 2022-08-02:
We feel that this is completed in new TopoStats

closed 2022-08-02 by SylviaWhittle