AFM-SPM/TopoStats #45: Nanoscope.txt file used in pygwytrcing not in repository

derollins, 2021-08-02:
The colour gradient file used in the code isn't in available in the repository and there is no instructions on how to install it.

alicepyne, 2021-08-02:
The nanosocpe.txt file should be added to the repository and should then be placed in  ~/.gwyddion/gradients/
[Nanoscope.txt](https://github.com/AFM-SPM/TopoStats/files/6917189/Nanoscope.txt)

MaxGamill-Sheffield, 2022-02-15:
3 ways to solve in the docker implementation:
1) Download the Nanoscope.txt from Alice's answer above, move it into the Docker-connected TopoStats folder then `mv /home/TopoStats/Nanoscope.txt /usr/share/gwyddion/gradients/nanoscope.txt` to match the config file.
2) Do 1) but pre-loaded onto the docker instance.
3) Simply change the necessity for the Nanoscope.txt file and change the config.ini file to 'Gywddion.net' where 'nanoscope.txt' is referenced

bobturneruk, 2022-04-01:
This is the specific colour gradient used to represent AFM data.

bobturneruk, 2022-04-06:
@benthomasuos will this be sorted by doing #84, please?

benthomasuos, 2022-04-06:
Yes, by removing Gwyddion from all the saveFile functions we can put in the custom matplotlib colormap that replicates the builtin Gwyddion.net gradient. This will remove the requirement for the Nanoscope.txt file entirely. However it won't solve the current issue of the file not being available for Docker at the moment.

bobturneruk, 2022-04-06:
Thanks @benthomasuos - I'll move this off our milestone.

bobturneruk, 2022-04-07:
On `master` would could solve this by editing the docker file.

ns-rse, 2022-04-08:
I've spent some time looking at this this morning and thought I'd document what I have tried.

1) Place `Nanoscope.txt` under `<path-to>/TopoStats/.gwyddian/gradients/Nanoscope.txt` _before_ starting the Docker container/image with the directory mounted. This means within the running image I can see the file before any processing.
2) Also copied the file to `/usr/share/gwyddion/gradients/nanoscope.txt` as suggested (also tried with `/usr/share/gwyddion/gradients/Nanoscope.txt` and `/usr/share/gwyddion/gradients/Nanoscope`.

The resulting image looks like...

[image]

...which I do not think is quite right, the colour scale is meant to range from black at `0.0` to white at the upper limit (`3.0`) if I understand correctly (@Jean-Du supplied a copy of the colour bar via Slack).

I've started looking for where `Nanoscope.txt` is used in the current `HEAD` of the `master` branch and its mentioned in the following files (output from `grep` indicates the matching `<filename>:<line>:` at the start of each match)...

```bash
❱ grep -R nanoscope -i -n            
config.ini:49:; Save file name for output colour scale eg: Nanoscope.txt
config.ini:50:savefile_zscalecolour = Nanoscope.txt
README.md:8:This search also excludes any files of the format ‘_cs’ which are cropped files exported by the Nanoscope Analysis software.
pygwytracing.py:590:    palette = data.set_string_by_name("/" + str(k) + "/base/palette", "Nanoscope.txt")
pygwytracing.py:625:    palette = data.set_string_by_name("/" + str(k) + "/base/palette", "Nanoscope.txt")
pygwytracing.py:666:        palette = data.set_string_by_name("/" + str(i) + "/base/palette", "Nanoscope.txt")
default_config.ini:49:; Save file name for output colour scale eg: Nanoscope.txt
default_config.ini:50:savefile_zscalecolour = Nanoscope.txt
```

So its present in the `[default_]config.ini` files and `pygwytracing.py`.

In `pygwytracing.py` it is mentioned under functions `saveunknownfiles()` (lines 571-603) and `savefilesasgwy()` (lines 606-636) which both appear to save copies of the file (although in the runs I've done these are not created).

In the `config.ini`/`default_config.ini` file it only sets the name to be used in saving these files and its stored in the variable `savefile_zscalecolour`.

This variable is then referenced in `pygwytracing.py` in a few places...

```bash
❱ grep -R savefile_zscalecolour -n -i
config.ini:50:savefile_zscalecolour = Nanoscope.txt
pygwytracing.py:552:    palette = data.set_string_by_name("/%s/base/palette" % k, savefile_zscalecolour)
pygwytracing.py:764:    savefile_zscalecolour = str(config.get("ImageOutput", "savefile_zscalecolour"))
pygwytracing.py:765:    print("Save file name: " + str(savefile_zscalecolour))
default_config.ini:50:savefile_zscalecolour = Nanoscope.txt
```
Lines 764 and 765 of `pygwytracing.py` are where this is loaded from the configuration files (and it also matches the two configuration files unsurprisingly), but it doesn't appear to load the palette in anyway, only gets the filename (`Nanoscope.txt`) and uses it.

I'm wondering if the machines where this palette is correctly used are at a different commit or branch. :thinking: 

I've also spent some time working out how to update a the Docker image with this file, but that is somewhat irrelevant since the images are not yet correctly produced.

alicepyne, 2022-04-08:
There are definitely some issues with consistent use of colour scales in TopoStats.

As I remember the “working version” had pygwytracing.py save out files in the nano scope colour format when specified. This is done at:
L903 savefiles(data, filename, extension)
894 # savecroppedfiles(path, data, filename, extension, orig_ids, crop_ids, minheightscale, maxheightscale)

The traced images are outputted in Matplotlib colour scales (what you show in your example), not by design but by default! It would be nice to have the option to make all of these the same.
L879 dna_traces.saveTraceFigures(filename, channel_name, minheightscale, maxheightscale, 'processed')

I hope this is helpful, but apologies if I’ve
Missed the point or made this more confusing.

⚠ Caution: External sender

[image]<https://user-images.githubusercontent.com/17566406/162418342-0240c9b4-50d4-4715-b920-b82f666555bf.png>

...which I do not think is quite right, the colour scale is meant to range from black at 0.0 to white at the upper limit (3.0) if I understand correctly ***@***.***<https://github.com/Jean-Du> supplied a copy of the colour bar via Slack).

MaxGamill-Sheffield, 2022-11-15:
I believe this is solved with the nanoscope.txt file converted to a matplotlib colourmap in `themes.py`

closed 2022-11-15 by MaxGamill-Sheffield