AFM-SPM/TopoStats #79: User Guide / Instructions

bobturneruk, 2022-03-28:
Considerable progress has been made on [installation instructions](https://github.com/AFM-SPM/TopoStats/issues/10), but we still lack a good user guide. 😢

Useful basis material here:

- [Guide.on.pygwytracing.pdf](https://github.com/AFM-SPM/TopoStats/files/8364989/Guide.on.pygwytracing.pdf)
- https://www.sciencedirect.com/science/article/pii/S1046202321000207

The user instructions might ultimately live in readthedocs which is underway for our [`dev`](https://github.com/AFM-SPM/TopoStats/tree/dev) branch: https://topostats.readthedocs.io/en/dev/?badge=dev

The instructions should explain the parameters in the config files https://github.com/AFM-SPM/TopoStats/blob/master/default_config.ini

ns-rse, 2022-12-01:
We have good documentation in place now (see [GH Pages](https://afm-spm.github.io/TopoStats/)) that covers installation and basic usage and most of the refactored API. Could still do with bulking out what is going on in the background under Workflow, but I'd say most things are covered and we could reasonably either close this issue or keep it as an "Epic" and break outstanding improvements into smaller tasks and link to them from here?

What do others think?

ns-rse, 2023-02-28:
This is related to #242 and there are some Notebooks in place already see #481 they don't yet work in Google Colab/Binder just yet (had issues loading SPM files into Notebooks locally).