AFM-SPM/TopoStats #87: Config file - what parameters should we capture?

bobturneruk, 2022-04-01:
Please put ideas here!

Jean-Du, 2022-04-01:
A separate config file for the Plotting.py

benthomasuos, 2022-04-04:
A plot theme or at least a default matplotlib colourmap with traditional AFM colourscale?

LunchBobs, 2022-04-22:
I've been writing an ideal config file here https://drive.google.com/drive/u/0/folders/1_j4PQH0a2NxKl69coTfZ4GyXGz5CS_gr with some ideas based on what the industrial collaborators need. I've tried to comment as much as I can so it makes sense.

bobturneruk, 2022-04-24:
Can you upload here please, @LunchBobs? This would enable more people to see and contribute...

LunchBobs, 2022-04-25:
Sure! [Ideal pygwytracing config file.docx](https://github.com/AFM-SPM/TopoStats/files/8552330/Ideal.pygwytracing.config.file.docx)

ns-rse, 2023-10-13:
Closing, we have a well-developed configuration file.

closed 2023-10-13 by ns-rse