AFM-SPM/TopoStats #92: Standardise nomenclature for branches in contributing guidelines

ns-rse, 2022-04-04:
It can be overwhelming and at times confusing about what branches exist for what purpose and who they are created by.

To help alleviate this standardising the nomenclature used in creating branches to be informative can be useful and documenting this in the [`contributing.md`](https://github.com/AFM-SPM/TopoStats/blob/dev/contributing.md) would be appropriate.

One approach to branches is to use the structure of `<username>/<github-issue>-<concise-description>` which tells anyone...

1. Who created the branch based on their GitHub username (`<username>`).
2. What GitHub issue ot pertains to (`<github-issue>`).
3. Some information about the issue.

As an example if I were to create a branch to address this issue I would call it `ns-rse/##-contributing-nomenclature`. Long and verbose branch names should be an issue when using `git` at the command line since it includes auto-complete via `Tab`.

This is just one example of standardised nomenclature and is meant to serve as a starting point for discussion before being the settled upon.

bobturneruk, 2022-04-05:
I think this is a good plan, although we do have people doing quite experimental stuff in the repo that probably will never be merged without significant refactoring / cherry picking. Should we recommend forks for this?

ns-rse, 2022-04-05:
Yes, I think if people fork the repository that would work well, its simple enough to make pull requests from forks (I did so this morning on [pySPM](https://github.com/scholi/pySPM/pull/25) as it was giving warnings).

ns-rse, 2023-04-24:
Closing main contributors have adopted the practice, for the time being I don't feel it needs codifying. Perhaps if we start getting external contributors we can revisit.