AFM-SPM/TopoStats #95: Filters.py/row_col_quantiles() : Binary_mask argument detection not working

SylviaWhittle, 2022-04-04:
In `filters.py` , in the function `row_col_quantiles(image: np.array, binary_mask: bool = False)`, the binary mask checking no longer works, since later on, the script checks if a mask was provided using: if `binary_mask:`, which fails if `binary_mask` is of type `np.ndarray`. I am not sure how to solve this, as this argument needs to be able to take two types of argument, and then the script needs to be able to detect if a value is passed with the function for that argument versus if it has not. The previous solution I had for this was to use: `… , binary_mask = None`, along with `if binary_mask is not None`. We could use type checking to see if the `binary_mask` argument is of type `np.ndarray` , however this would still not allow for good type-hints. 

We could use `Unison()`  to allow for more than one type to be hinted, however this requires an import and may not be best practice.

bobturneruk, 2022-04-05:
@ns-rse - Is this something you can work on, or do you need more info?

ns-rse, 2022-04-05:
Sorry for not clocking this earlier.

Apologies for the mistake here, I can fix this. Where abouts is the failure occurring with `binary_mask` being an `np.ndarray` @SylviaWhittle ?

I'd noticed the `binary_mask` was used in a few different functions and it's one of the reasons I thought that converting this to a class would be useful here. If an object is binary (which it is here since it is tested for being `None` or anything else) then its best to have it as a `boolean`.

I've been working on the refactoring `filters.py` functionality to a class and starting to get things working on the multiprocessing and just pushed my work so far (see [`filters.py`](https://github.com/AFM-SPM/TopoStats/blob/ns-rse/90-refactor-filters/topostats/filters.py) and [`topotracing_v2.py`](https://github.com/AFM-SPM/TopoStats/blob/ns-rse/90-refactor-filters/topostats/topotracing_v2.py)), but clearly I have to get the logic correct so I'll go back and make sure I do that.

ns-rse, 2022-04-05:
Thanks @SylviaWhittle I'll have a go at running it and investigate and let you know how it goes.

ns-rse, 2022-04-05:
Ok, I see whats happening now and I'd not clocked everything on first look, this is essentially my notes as I walk through understanding it and fixing my mistake. I recreated the error and get...

```(topostats) ❱ python topostats/topotracing.py 
2022-04-05--18-12-15
Traceback (most recent call last):
  File "/home/neil/work/projects/topostats/TopoStats/topostats/topotracing.py", line 75, in <module>
    data_second_flatten = filters.align_rows(data_initial_flatten, binary_mask=mask)
  File "/home/neil/work/projects/topostats/TopoStats/topostats/filters.py", line 62, in align_rows
    row_quantiles, _ = row_col_quantiles(image, binary_mask)
  File "/home/neil/work/projects/topostats/TopoStats/topostats/filters.py", line 30, in row_col_quantiles
    if binary_mask:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

```

Then in `topotracing.py` there is...

```
mask = data_initial_flatten > threshold
```

...and this compares every value in `data_initial_flatten` and returns a `np.ndarray` as you say. It looks like this...

```
binary_mask : [[ True False False ... False False False]
 [False False False ... False False False]
 [False False False ... False False False]
 ...
 [False False False ... False False False]
 [False False False ... False False False]
 [False False False ... False False False]]
<class 'numpy.ndarray'>
```

It The `mask` Numpy Array is then used to determine whether masking is applied in the second call to `filters.aign_rows()` with...

```
data_second_Flatten = filters.align_rows(data_initial_flatten, binary_mask=mask)
```
...and because it will _always_ be an Numpy array masking is _always_ applied at this stage.

This raises a question for me, should we _always_  mask on second flattening? Or is it conditional on whether all values in the array are above the threshold?

If we _always_ mask on the second flattening and the fact the code initially meant this was always going to evaluate to `True` and the comment at the start of the block of code is `# Masked flattening` also suggests this, then its simply a case of _not_ using the `mask` array but instead using `binary_mask=True` in the second flattening...

```
# Masked flattening
logging.info('masked flattening')
logging.info('masked align rows')
data_second_flatten = filters.align_rows(data_initial_flatten, binary_mask=True) # True instead of mask
plottingfuncs.plot_and_save(data_second_flatten, 'plot_data/masked_align_rows.png')
logging.info('masked x-y tilt')
data_second_flatten = filters.remove_x_y_tilt(data_second_flatten, binary_mask=True) # True instead of mask
plottingfuncs.plot_and_save(data_second_flatten, 'plot_data/masked_x_y_tilt.png')
```

The other possibility, although I don't think this is what is the desired functionality, is that the second flattening only uses masking if _all_ values after the initial flatten are `< threshold`. This would mean that every value in the `mask` Numpy Array is `False`. 

Python is being quite helpful here because its telling us in the `ValueError` message what we should use to get the desired result and that is [`a.any()`](https://numpy.org/doc/stable/reference/generated/numpy.any.html) or [`a.all()`](https://numpy.org/doc/stable/reference/generated/numpy.all.html) (this isn't always the case with Python though, some error messages are very cryptic!).

If we wanted to do masking conditional on _all_ values in the array being `< threshold` then we can use `a.any()` to tell us if there is one or more value in `mask` that is `True`, so after doing the assessment we can check this and set `mask` to be a boolean and then it (will hopefully!) "just work".

```
# Thresholding
logging.info('otsu thresholding')
threshold = filters.get_threshold(data_initial_flatten)
logging.info(f'threshold: {threshold}')
mask = data_initial_flatten > threshold
print(f'What is the truthfulness of mask.any()? : {mask.any()}') # For debugging
mask = mask.any()
print(f'Do we have values that are > threshold? : {mask}') # For debugging
plottingfuncs.plot_and_save(mask, 'plot_data/binary_mask.png')

# Masked flattening
logging.info('masked flattening')
logging.info('masked align rows')
data_second_flatten = filters.align_rows(data_initial_flatten, binary_mask=mask)
plottingfuncs.plot_and_save(data_second_flatten, 'plot_data/masked_align_rows.png')
logging.info('masked x-y tilt')
data_second_flatten = filters.remove_x_y_tilt(data_second_flatten, binary_mask=mask)
plottingfuncs.plot_and_save(data_second_flatten, 'plot_data/masked_x_y_tilt.png')
```

All that said I think what is required is the former of _always_ performing Masked flattening, since in the original code `if binary_mask is not None` would always evaluate to `False` with `mask = data_initial_flatten > threshold`.

I hope that makes sense (and that I've understood what is going on here), if you could let me know which desired behaviour is desired...

1. Always flatten a second time with masking.
2. Only flatten a second time if one or moreof the values after the initial flatten are `> threshold`.

...then I can make the changes in this branch so its working again and apply the correct logic in my refactoring.

SylviaWhittle, 2022-04-05:
That indeed makes a lot of sense @ns-rse! Thank you so much for looking into this. Yes, your first solution seems good since we will always want to mask on the second function call. I would like to ask about passing the mask into the function. Will the function have access to the mask if it isn't passed into the function, and if not, shouldn't we strive to keep it being passed into the function regardless for easier code readability?

ns-rse, 2022-04-05:
Thanks for clarifying that @SylviaWhittle I've made that change and will push it shortly.

> Will the function have access to the mask if it isn't passed into the function, and if it will, shouldn't we strive to keep it being passed into the function regardless for easier code readability?

Because the changes I made set a default for the `mask` value for all three functions `row_col_quantiles()`, `align_rows()` and `remove_x_y_tilt()` with...

```
def row_col_quantiles(image.np.array, binary_mask: bool=False) -> np.array:
...
def align_rows(image.np.array, binary_mask: bool=False) -> np.array:
...
def remove_x_y_tile(image.np.array, binary_mask: bool=False) -> np.array:
...
```

...then by default its value will be `False`, unless its explicitly set to `binary_mask=True` on calling the function.

Another approach would be to _not_ set a default value, and then the code in `topotracing.py` will _always_ have to explicitly state whether `binary_mask` should be `True` or `False`.

This is something I'd not clocked in my refactoring though so I'll have to tweak what I've done to account for this but its not too big a change. I'll also go through `topotracing.py` more carefully to understand what the desired behaviour is and get it correct.

Thank you very much for your help @SylviaWhittle, I really appreciate and need your guidance with this :+1:

SylviaWhittle, 2022-04-05:
Ah thanks for the clarification @ns-rse - though I think I may not have explained my question well enough - I meant to ask that since the functions now just take a boolean value for the `mask`  argument (to tell the function whether to use the mask or not), the binary mask that is obtained in `topotracing.py` will not be passed to the functions in `filters.py`. I was wondering if this means that the functions in `filters.py` will be able to 'see' the mask that is being used in `topotracing.py`?

TLDR: How will functions like `align_rows` know what the mask's contents are if we don't pass the mask in the function call in `topotracing.py`? (It's possible I am misunderstanding you though!)

ns-rse, 2022-04-05:
It could just as easily be me who is misunderstanding, but the way the code is structured just means that we have to be explicit when calling the functions defined in `filters.py` from `topotracing.py` as to whether we want to mask or not.

Because `mask = data_initial_flatten > threshold` _always_ results in "something" then the previous test of `if binary_mask is not None` when passed `binary_mask=mask` will _always_ be true, since `mask` is an object rather than `None`.

If the process is always...

1. Flatten without masking
2. Flatten with masking
3. Zero the average background with masking (just clocked this)

Then we just need to be explicit at each step as to whether to apply masking or not when calling the functions, whether that is `align_rows()`, `remove_x_y_tilt()` or `row_col_quantiles()`.  Each of these functions is written because we re-use it, but how it is used/applied depends on the logic/arguments they are given (in this case `binary_mask` and whether it is `True` or `False`).

Originally this `True`/`False` assessment was based on whether the initial flattening had been performed, but if we always want to do these steps then we can just be explicit and say so in the way we call the functions.

ns-rse, 2022-04-06:
After a nights sleep and with fresh eyes I've now seen what I've done wrong and what you've been getting at and its completely my misunderstanding.

The masking is done with `np.ma.masked_array(image, mask=binary_mask, fill_value=np.nan)` (within the `row_col_quantiles()` function. I've never masked an array before so have gone and read up what this is doing (at [NumPy : Masked Array - What is a masked array](https://numpy.org/doc/stable/reference/maskedarray.generic.html#what-is-a-masked-array) and now I understand why `binary_mask` has to be a `np.ndarray` because it doesn't just control the logic of whether to apply `np.ma.masked_array()` it is required to say which data points within the array need masking with `NA`.

I've reverted the typehints and calls back to the original and updated pull request #97 although I can't work out how I've managed to include all the previous commits with notebooks.

I'm so sorry to have wasted your time with this and will endeavor to be more careful and ask in the future if I don't understand something.

ns-rse, 2022-05-19:
This has been addressed in #99

closed 2022-05-19 by ns-rse