AFM-SPM/TopoStats #101: Handling tip convolution errors in grain sizing

bobturneruk, 2022-04-07:
**Is your feature request related to a problem? Please describe.**
We are working with shapes that approximate to the outline of simple geometric objects e.g. the outline of a square. Depending on the size of the AFM tip shapes appear larger or smaller. This can be corrected experimentally, but we'd like to be able to compare data across different tip sizes.

[image]

Small tip left, big tip right.

**Describe the solution you'd like**
Fitting the outline of the shape at it's highest point (like a ridge running round the shape) might give us a true object size that's independent of tip size.

[image]

**Describe alternatives you've considered**
Another approach would be deconvolution, but this would maybe need the tip size which we don't generally know.

bobturneruk, 2022-04-07:
This would need someone familiar with topostats code for tracing loops of DNA to develop further.

LunchBobs, 2022-04-22:
@bobturneruk I've accidentally added a duplicate of this issue, but put a bit more detail in. Is it possible to delete this one?

bobturneruk, 2022-04-24:
Sure @LunchBobs - I'll just close this and mention #108 so we have a record that they're linked.

closed 2022-04-24 by bobturneruk