AFM-SPM/TopoStats #102: Processed Images Contain a Border of 0's

MaxGamill-Sheffield, 2022-04-11:
**Describe the bug**
- When TopoStats processes the images, a single pixel border is added to the image array which has a value of 0.
- i.e. For a 1024x1024px image, the processed tiff without any scale bars/rulers is 1026x1026px. 512x512px spm's become 514x514px

**To Reproduce**
- Run TopoStats as usual (xvfb-run python pygwytracing.py) with the rulers and scale bars off in the config file.
- Config file attached (as .txt as .ini not supported) 
[config.txt](https://github.com/AFM-SPM/TopoStats/files/8463856/config.txt)

**Expected behavior**
- Output image dimensions should be the same as input image dimensions

**Output**
- Screenshot attatched showing difference in input and output size.
[image]

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - Docker used

bobturneruk, 2022-04-11:
Thanks for this @MaxGamill-Sheffield!

MaxGamill-Sheffield, 2022-04-11:
I think it's an issue with the gwy.gwy_file_save() function as the data is still the original size at all points in the pipeline, and the produced images immediately after import are larger.

I can work on a function to remove this.

MaxGamill-Sheffield, 2022-11-15:
resolved as no longer using gwyddion saving functions

closed 2022-11-15 by MaxGamill-Sheffield