AFM-SPM/TopoStats #108: Problem with tip convolution

LunchBobs, 2022-04-22:
**Is your feature request related to a problem? Please describe.**

An issue I have when analysing closed loop RNA constructs is that as the tip radius changes, so does the projected area of the molecules. This makes it really hard for me to compare the statistical quantities like mean area and mean radius between imaging session (and sometimes in the same session). Think of it like this;

Draw the same size square with a thin pen and a board marker (green)

[image]

You drew the same shape, but if you calculate the projected area covered by the pen the fat marker image is much larger (yellow).

[image]

All the molecules we look at are made of the same stuff, and can't really be fatter or thinner. RNA doesn't put on weight. Our AFM tips act like different size pens, and as we can't always get the same tip radius (we try to!) some days the molecules look fat, other days they look thin. This makes it difficult to compare the area distributions from different days and tips (sometimes even the same tip can get fatter over time). The axes on these mock graphs would be projected area on the X axis, and number of molecules on the Y.

[image]

**Describe the solution you'd like**

What we'd like to be able to measure would be something I'm going to call the contour area. If we trace around the highest point of the molecule (or the other skeletonising work that topostats has) we could measure the area inside. Step one would be to trace the outline of the molecule (black line on the green pen lines)

[image]

Then we calculate the area inside the contour loop (yellow)

[image]

These would be the same, irrespective of the pen size/ AFM tip radius, and the area distributions would be comparable. The axes on these mock graphs would be projected area on the X axis, and number of molecules on the Y.

[image]

ns-rse, 2022-05-16:
One solution for dealing with this sort of problem is to take a skeleton as suggested in [this thread on image.sc](https://forum.image.sc/t/transform-area-roi-into-a-skeleton-line-roi/67070).

Scikit-Image has a [skeletonize](https://scikit-image.org/docs/stable/auto_examples/edges/plot_skeleton.html) image that could be tested.

A quick test shows it's perhaps not quite what we are after, I used the following code which creates a very crude and simple square with a hole in the middle, plots it, skeletonize's it and then plots it again...

```python
import numpy as np
import matplotlib.pyplot as plt
from skimage.morphology import skeletonize

square_with_hole = np.array(
        [
            [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 1, 1, 1, 0, 0, 0, 1, 1, 1, 0],
            [0, 1, 1, 1, 0, 0, 0, 1, 1, 1, 0],
            [0, 1, 1, 1, 0, 0, 0, 1, 1, 1, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0],
            [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
        ]
)
def plot_array(data,
               interpolation: str = 'nearest',
               cmap: str = 'afmhot',
               out: str = 'test.png'):
    fig, ax = plt.subplots(1, 1, figsize=(8, 8))
    ax.imshow(data, interpolation=interpolation, cmap=cmap)
    plt.savefig(Path(out))
    plt.close()
    return fig, ax

## All points
plot_array(square_with_hole, out='all_points.png')

## Skeletonize
skeleton = skeletonize(square_with_hole)
plot_array(skeleton, out='skeleton.png')
```

The original image looks like...

[image]

The skeleton looks like...

[image]

So it works with this crude example of a "loop" as well as the documented examples of more linear objects.

ns-rse, 2022-05-19:
I've started looking at the `topostats/dnatracing.py` and `topostats/tracingfuncs.py` and it looks like the skeleton method or a variant of it is already in place and being used.

`topostats/dnatracing.py` line 126, which is commented out, calls scikit-image's `skeletonize()` method. Instead the `topostats/tracingfuncs.py` classes are used which implements [A fast parallel algorithm for thinning digital patterns (1984) Zhang TY, Suen CY](https://dl.acm.org/doi/10.1145/357994.358023).