AFM-SPM/TopoStats #111: Metadata file output

LunchBobs, 2022-04-22:
**Is your feature request related to a problem? Please describe.**
Analysis performed is a bit of a black box. There's no way to determine what parameters were used when going back to data previously analysed (unless meticulous notes were kept at the time). 
- No list of functions performed and parameters used
- No option to select which functions are used
- Some images are not analysed, but don’t know which ones

**Describe the solution you'd like**
A clear tree of functions that were performed on the data, in the order they were performed and what parameters were used will be great. The file should also list the names of the images analysed, those that were not analysed and the number of images analysed in total. It could also function as a log of any errors the software encountered during processing.

**Describe alternatives you've considered**

It might be possible to copy the config file as a metadata file into the folder that it puts the processed images in as a short term solution. It wouldn't have everything you'd want but at least there'd be the values you'd used in the config in there. Nanoscope analysis has a visual version of this onscreen when you open the image, that might be one way to do it. 

**Additional context**
Add any other context or screenshots about the feature request here.

benthomasuos, 2022-05-20:
I've had a think about this alongside what could be done to open up the data formats from Bruker's incomprehensible binary formats.
HDF5 files could store the raw data directly from the numpy arrays that PySPM generates. The config data could be stored in these files as well as particular settings for each function that is run on the data. It could also store the analysis/grain stats in a standardised format for subsequent labelling and meta-analysis. The resultant ".h5" files could then be shared amongst AFM researchers and used with other programming languages other than Python, e.g. R or Matlab

The only downside is that HDF5 isn't a text format so you can't just open up the file and view the config or metadata in a program like notepad. We would need to create some utility functions to cope with that so it would either output the config and/or metadata to a text file or just to the command line. There are HDF5 GUI programs, which could be used for viewing the data too.

Creating a HDF5 intermediary file could sit in the current `dev` branch workflow between reading the .spm file with PySPM and the first action of filters. It could be a class that provides helpful get/set methods like `.get_height_array()` `.get_grain_labels()` and `.set_height_array()` `.set_grain_labels()`

bobturneruk, 2022-05-23:
Thanks @benthomasuos - here's a riff on that, it's hack, let's say it's pseudo code:

```{python}
import re

import h5py
import numpy as np
from pySPM.Bruker import Bruker

minicircle = Bruker('minicircle.spm')

with h5py.File("minicircle.hdf5", "w") as f:
    for metadata, value in minicircle.scanners[0].items():
        f.attrs.create(metadata, value)

    for layer in minicircle.layers:
        layer_name = re.search("\"(.+)\"", str(layer[b"@2:Image Data"])).group(0).strip("\"")
        layer_data = minicircle.get_channel(layer_name)
        layer_shape = np.array([layer_data.size['pixels']['x'], layer_data.size['pixels']['y']])
        dset = f.create_dataset(layer_name, layer_shape, dtype='f')
        dset[:, :] = layer_data.pixels
        for metadata, value in layer.items():
            dset.attrs.create(metadata, value)

```

[image]

I think this might sit better in a separate repository as a metadata proposal and Python API, to aid reuse beyond topostats. We might want to consider json-ld as a text based metadata-only adjunct to this. Maybe the API could handle both.

ns-rse, 2022-06-16:
In the interim I've added a function `write_yaml()` and included a call to it at the end of `run_topostats.py` which writes the configuration to YAML in the output directory (it was relatively straight-forward so figured it would be useful anyway as it keeps the config with the output).

ns-rse, 2023-01-06:
Revisiting this.

The configuration used for processing is written to the output as suggested.

However, before recording _exactly_ what steps were undertaken and in what order the process does I feel need documenting up-front in the first place.

To this end I started ages ago trying to document what theses steps actually are (in no small part to aid my understanding of the process when it comes to working with the code). These are in a crude flow diagram in the [documentation workflow](https://afm-spm.github.io/TopoStats/workflow.html). However, I feel these could be improved on if we improved them and adopted the [Unified Model Language](https://en.wikipedia.org/wiki/Unified_Modeling_Language) which is specifically designed for such documentation and has things such as Class Diagrams which are perfect for documenting the properties of our classes and how they interact (for a very simple example of these in use see the diagram under [this section](https://realpython.com/inheritance-composition-python/#composition-in-python)).

ns-rse, 2023-11-10:
I still think we need to improve the [workflow documentation](https://afm-spm.github.io/TopoStats/workflow.html) to show the different steps and what they mean.

However, are people happy with having the configuration options written to YAML?

Is the `.topostats` file recording metadata as suggested above @SylviaWhittle ?