AFM-SPM/TopoStats #112: dev-remove-gwyddion: Automate dx

MaxGamill-Sheffield, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
dx value is currently taken as a user input / hard coded. This can be automated from the image metadata.

**Solution**
Once channel is chosen from the spm data, e.g. "height = scan.get_channel", you can use the following to automate getting dx:
```
extent = height.get_extent()
dx = extent[1]/len(height.pixels) #image extent in nm / pixel dimension of img
#similarly for dy
dx = extent[3]/len(height.pixels) #image extent in nm / pixel dimension of img
```
This assumes the image starts from 0 (a.k.a. that extent[0] and extent[2] == 0) which the extent usually does but may not with non-square images - this would need to be tested.

bobturneruk, 2022-04-28:
Cheers @MaxGamill-Sheffield - for future reference, how is `dx` defined, please?

SylviaWhittle, 2022-05-13:
This has now been implemented.