AFM-SPM/TopoStats #113: dev-remove-gwyddion: Gaussian filter sigma - nm or px?

MaxGamill-Sheffield, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
Main branch TopoStats uses a fixed gaussian filter size of 1.5px. Currently this is fixed as a user input / hard coded variable in this branch and is also in nm.

**Describe solutions and alternatives**
We can either hardcode this as 1.5px like before or continue using nm which then gets converted to pixels but might not be best when you have different scales of the same sample - would need testing.

Is this a worthwhile debate as to which might be best?

bobturneruk, 2022-04-28:
My intuition is that a default in physical units might be best.

SylviaWhittle, 2022-05-13:
This may be a question for @alicepyne.

alicepyne, 2022-05-13:
I think nm (physical units) is best!

ns-rse, 2022-12-01:
We've dealt with nanometer/pixel in the refactoring to remove Gwyddion dependency, other relevant PRs...

#121
#131 
#212 
#272

closed 2022-12-01 by ns-rse