AFM-SPM/TopoStats #114: dev-remove-gwyddion: Remove small objects

MaxGamill-Sheffield, 2022-04-28:
**Describe the bug**
Remove small objects doesn't seem to be working in the dev-remove-gwyddion branch, but the same value in the main branch TopoStats can remove the objects.
(Talk that in the main branch TopoStats, the specified value is a length not an area)

**To Reproduce**
Run using the minicircle.spm file and parameter config settings below (some will need to be added after 'Height' channel is extracted:
gaussian_size = 1.5 * dx # makes gaussian 1.5 px (like main branch TS) instead of nm
dx = height.get_extent()[1] / len(height.pixels) # this automates the px-to-nm 
upper_height_threshold_rms_multiplier = 1.3
lower_threshold_otsu_multiplier = 0.7
minimum_grain_size_nm = 300

**Expected behaviour**
The 4 small grains seen in the bounding box figure should be removed like they are in current TopoStats. It might be a simple fix of increasing the value but why is this not consistent between main branch and this branch? [Gwyddion](http://gwyddion.net/documentation/head/pygwy/gwy.DataField-class.html#grains_remove_by_size) also takes in the pixel area but manages to remove them although the source code is not helpful in determining this.
I think these relate to grains: 8, 11, 19, 23 and have pixel areas of 777, 778, 653, 911 in the dev-remove-gwyddion branch.

These extra grains are found when `minarea = 110e-9` in the main branch TopoStats.
I think this relates to grains: 9, 12, 20, 24 and have pixel areas of 774, 773, 642, 886. So perhaps it is this one which is broken? 

**Output**
[image]

Jean-Du, 2022-04-28:
For clarification, in the main branch, the parameter "minarea" (which is actually a length) is only used in the grainfinding function and not the removesmallobjects, so only grains larger than minarea are picked up in the first place; the removesmallobjects and removelargeobjects both remove things in relation to the median grain size obtained after grainfinding; they use the parameters mindeviation and maxdeviation (for example, if mindeviation=0.5 and maxdeviation=1.5, only grains whose size fall into the range between 0.5*median and 1.5*median are retained after removesmallobjects and removelargeobjects)

MaxGamill-Sheffield, 2022-04-28:
@Jean-Du Ah ok gotcha. Do you know what length is it? The max possible length of the molecule?

Jean-Du, 2022-04-28:
@MaxGamill-Sheffield So the filtering in grainfinding is still based on area, but the parameter "minarea" is actually a length. What the main branch currently does is that, the "minarea" (a length) is divided by the length of a pixel to obtain the number of pixels, and that number is used as the selection criteria for masking--only grains whose area is larger than that number are masked. 
This can probably be fixed by changing the minarea into an actual area, and divide that over the size of a pixel instead of its length, but that would require different values.

MaxGamill-Sheffield, 2022-04-28:
@Jean-Du Ah ok, so the length is converted to a number of pixels, and then grains that spread across less than that number of pixels are not masked. That seems like what is currently happening in this branch so I don't understand why the results are different:
[image]

Jean-Du, 2022-04-28:
@MaxGamill-Sheffield  Yes, the removesmallobjects removes things based on the parameter "mindeviation"; it multiplies this with the medial grain area and uses that as the threshold, not the minarea.
To confirm this is the issue, you can set the "mindeviation" to 0 for the old TopoStats, and you will see that those small grains won't get removed, as shown here
[image]

MaxGamill-Sheffield, 2022-04-28:
Yep! Just ran it myself and I see the exact same thing as in the dev-remove-gwyddion branch!

Suggestion to mirror the main branch TopoStats and add a new parameter of 'min_deviation' to use with the median pixel area of the found grains (after removal of those < minsize) and remove grains below min_deviation * median pixel area.

SylviaWhittle, 2022-08-02:
Fixed ✅

closed 2022-08-02 by SylviaWhittle