AFM-SPM/TopoStats #116: dev-remove-gwyddion: Add yml file dependancy install

MaxGamill-Sheffield, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
As this doesn't need docker, there is no dependancy list to get this working as a stand alone, nor is there instructions.

**Describe the solution you'd like**
I've made a yml file of my conda environment which runs this branch and can be installed using:
`conda env create -f environment.yml`
[environment.txt](https://github.com/AFM-SPM/TopoStats/files/8582724/environment.txt)
Added as a txt file because GitHub doesn't allow yml file upload

**Describe alternatives you've considered**
An environment dependancy list which doesn't use conda.

MaxGamill-Sheffield, 2022-04-28:
@ns-rse that's sounds great and very useful! 

Mind if I keep this issue open for now while I'm doing some quick testing just so I can get a dev environment going quickly with conda and this package list, and I'll close it down once your branch has been merged?

ns-rse, 2022-04-28:
@MaxGamill-Sheffield sure not a problem keeping this open at all, its useful to track it as a specific issue until such time as its been merged. :+1:

bobturneruk, 2022-04-29:
Great to see this interaction - I need to look at #103 but experience indicates that using conda for virtual environments, whilst using "standard" python tooling i.e. `setup.py` for installation is a good mix. For topostats I would be using conda to create an empty environment and then pip installing as @ns-rse suggests.

ns-rse, 2022-12-06:
Closing as we have `setup.cfg` listing all package dependencies and these are pulled in and installed when installing `topostats` be that via PyPI or from the git repository.

closed 2022-12-06 by ns-rse