AFM-SPM/TopoStats #118: Make method for thresholding flexible

ns-rse, 2022-04-28:
**Is your feature request related to a problem? Please describe.**

Currently the method used for thresholding is fixed using the [Otsu method]() as implemented in [skimage.filters.threshold_otsu](https://scikit-image.org/docs/stable/api/skimage.filters.html?highlight=otsu#skimage.filters.threshold_otsu).

This may not be the optimal/desired method and there are lots of other methods available (see those available at [skimage.filters](https://scikit-image.org/docs/stable/api/skimage.filters.html) for exampes.

**Describe the solution you'd like**

Making the method configurable with a default method that performs well in most instances is desirable.

This can be achieved using a [Factory Design Pattern](https://realpython.com/factory-method-python/) allowing additional filters to be added.

bobturneruk, 2022-04-29:
Hi @ns-rse - need to bear in mind @alicepyne 's input that having one approach that works for a wide range of input data so as to minimise the need for using tweaking is optimal. Nonetheless this sounds great and I'll have a look.

ns-rse, 2022-04-29:
Thanks @bobturneruk , I set the default to be that which is currently used `method='otsu'`. I figured adding this makes it easy to test alternative methods further down the line if that is desirable (documentation could even reflect that whilst other methods are available they may not be recommended).

skimage includes local thresholding which may be something to consider. Currently they can't be added with the workflow though as that would require adjusting how thresholds are applied to masking.

But that is getting way ahead, it was simple to add this and it starts making thresholding configurable rather than hard coded.

ns-rse, 2022-05-16:
Closed by #119

PR #119 (2022-05-13): 118 | Adds a factory method to make thresholding flexible
resolves #118 

This adds a simple [factory design pattern](https://realpython.com/factory-method-python/) to perform the threshold calculation on images.

It includes tests which show how to use it (and validate) and it should be easy to add additional methods.

To use it in `dev-remove-gwyddion` it would be necessary to [cherry-pick](https://rse.shef.ac.uk/blog/2021-04-23-cherry-pick/) commit `8f0542b` to get it into the branch (or possibly a `rebase` but I'm unsure how divergent `dev` and `dev-remove-gwyddion` are and that may cause  [... cut]
closed 2022-05-16 by ns-rse