AFM-SPM/TopoStats #122: Unit tests and refactoring of grainstats.py

ns-rse, 2022-05-06:
@SylviaWhittle has re-written the code that calculates grain statistics (see branch/file [`dev-remove-gwyddion/topostats/grainstats.py`](https://github.com/AFM-SPM/TopoStats/blob/dev-remove-gwyddion/topostats/grainstats.py)).

We need unit-tests in place and there may be a little refactoring that can be done at the same time.

Once complete it can be merged along with #103 and #118 into `dev` to form a basis for a working version in Python 3 that doesn't rely on Gwyddion.

SylviaWhittle, 2022-05-06:
The legitimacy of the calculations is not confirmed just yet. There have been some conflicts between the values that are computed in grainstats.py and Gwyddion, so there will probably be small changes to the calculations, though nothing major I think. @MaxGamill-Sheffield and I are looking into this. 

Should the unit tests and refactoring be done and then we add hotfixes for the calculation methods in the interests of progress and efficiency @ns-rse ?

ns-rse, 2022-05-09:
I'll write unit-tests with what is in place now, if the underlying calculations then change these target values we can modify the targets of the test to reflect that once the methods to be used are decided upon.

PR #136 (2022-05-16): Refactors grainstats.py and adds tests; corrects error in pixel scaling
Resolves #122 

I've refactored @SylviaWhittle `grainstats.py`, splitting out some code to small methods and written unit tests.

Had some merge conflicts with `dev` after merging #130 that I've now resolved in this branch.

I also noticed I had a mistake where I had divided (`/`) the grain size by the `pixel_to_nm_scaling` rather than multiplying (`*`), stupid error.

I've also applied [Black](https://github.com/psf/black) to many files which makes some files/changes appear greater than they are but many changes are cosme [... cut]
closed 2022-05-19 by ns-rse