AFM-SPM/TopoStats #126: Docs are not built correctly

bobturneruk, 2022-05-06:
**Describe the bug**
Docstrings should be built into HTML documentation here https://topostats.readthedocs.io/en/dev/index.html, but not all of this is being generated e.g. for `filters.py`.

**To Reproduce**
Visit https://topostats.readthedocs.io/en/dev/index.html

**Expected behavior**
All docstrings should be translated into HTML on readthedocs.

**Output**

Readthedocs log here: https://readthedocs.org/api/v2/build/16841145.txt

Or build locally.

**Additional context**
Add any other context about the problem here.

ns-rse, 2022-05-06:
Ah, I think I may be to blame here as I've used [Numpy Docstrings](https://numpydoc.readthedocs.io/en/latest/format.html) rather than reStructured Text.

I don't think its too hard to fix, it should be a case of adding a dependency for numpydoc to `docs/conf.py` so I'll have a look at this and try resolving it.

Sorry for messing things up.

bobturneruk, 2022-05-06:
I think if I've not fixed it I've got it OK for the time being via PR #127

ns-rse, 2022-05-06:
I was off the mark, missed that `sphinx.ext.napoleon` is already included in `docs/conf.py` so Sphinx was primed to already work with the NumpyDoc strings.

The breakage was however probably down to me though removing `setup.py` and using `setup.cfg` instead.

Builds locally, but I'm not familiar with how `.readthedocs.yaml` is parsed/used in GitHub Action as I see no reference to it in `.github/workflows/tests.yaml`, but perhaps its outside of workflows if a scan of [this](https://docs.readthedocs.io/en/stable/connected-accounts.html) suggests.

One thing I noticed is that the version of python in `.readthedocs.yaml` is pinned at `2.7`, is it perhaps worth removing that? I've commented it out locally and rebuilt the HTML documentation and it doesn't appear to have made any difference.

PR #127 (2022-05-06): Docs update
Resolves #126 

To test:

```
pip install -e ".[docs]"
cd docs
make html
```

Have a look at `_build/index.html` in a browser.

Readthedocs won't update until we merge so can't be 100% sure this will fix and another iteration may be needed.