AFM-SPM/TopoStats #132: dev-remove-gwyddion: Other Bruker formats not parsed

MaxGamill-Sheffield, 2022-05-11:
**Describe the bug**
Alternative to `.spm` files there are `.001` file formats too, and currently these are not analysed when using this branch.

**To Reproduce**
Run with any non-`.spm` AFM file.

**Expected behavior**
It should analyse these files.

**Additional context**
Should be able to be solved with by inserting the code below into the "# search recursively for files" section (there may be a nicer way to do it with glob in one line):
```
file_list1 = list(path_base.glob('**/*' + '.spm'))
file_list2 = list(path_base.glob('**/*' + '.[0-9][0-9][0-9]'))
file_list = file_list1 + file_list2
```

ns-rse, 2022-05-12:
In IAA meeting 2022-05-12 @alicepyne raised the need to support other formats.

As well as the above proposal this can be done relatively easily by extending the YAML configuration file to include a list of file types...

```yaml
file_ext:
  - .spm
  - .jpk
  - .other
```

Then in the code when we glob files to find them (currently in `topostats/utils/find_images`) with something similar to ([solution](https://stackoverflow.com/questions/47655205/pathlib-path-glob-and-multiple-file-extension))...

```python
    return list(base_dir.glob(f'**/*[{ext[0]}][{ext[1]}][{ext[2]}]' + file_ext))
```

...although some thought will need to be given to how to handle the length of the list of file extensions.

ns-rse, 2023-10-13:
@MaxGamill-Sheffield : Can we close this issue now? Is supporting the `.001` etc. file formats something that is still required?