AFM-SPM/TopoStats #137: Support loading and processing multiple file formats

ns-rse, 2022-05-13:
**Is your feature request related to a problem? Please describe.**
The ability to handle multiple file formats would be very useful.

**Describe the solution you'd like**
A single function or class to `io.load_scan(file_path)` that uses a [Factory Method](https://realpython.com/factory-method-python/) design pattern to use the appropriate functions/packages to load the file and return a NumPy array ready to be processed.

**Describe alternatives you've considered**
Currently we support Bruker formats using [pySPM](https://github.com/scholi/pySPM) but what other formats are commonly used and are there other Python packages out there that load these formats?

**Additional context**
At the same time this should address #132

bobturneruk, 2022-10-03:
Several High Speed AFM users have expressed an interest in using TopoStats - LibASD would enable High Speed AFM file formats to be opened, so might be a priority.

bobturneruk, 2022-10-03:
Should `io.py` hold some of the stuff that's currently in `filters.py`? Probably, yes. This means all the stuff that has to know about file format and headers is in one place and `io.py` is more of a self contained module.

MaxGamill-Sheffield, 2022-10-03:
e.g. a `pySPM` function is currently (`.get_channel`) being used in the filters class within the `extract_channel` function, this will probably break if loading via a different library as it will not be a `pySPM.Bruker` object.

I think the `Filters` class should have a universal input (image, px_to_nm and general config file stuff) and thus the `io` should be universal output of the image and anything other scripts need from the image headers.

ns-rse, 2022-10-04:
> can we point to examples of files we would like to be able to open? 

The `afmformats` package has sample files that it uses for tests, we could perhaps borrow those?

> I think also we may wish to create some subsidiary issues pertaining to specific file formats as we get closer to implementation.

Agree with this, I've created a specific issue for supporting JPK (see #300 ) and added it to the Milestone 3. I'll edit the top comment to have a list of issues for specific file formats.

> I think the Filters class should have a universal input (image, px_to_nm and general config file stuff) and thus the io should be universal output of the image and anything other scripts need from the image headers.

Agree with this whole-heartedly. Aside from the image (`np.ndarray` of heights) and the Pixel-to-nm scaling what other parameters might we want to extract across images consistently and pass onto the next step in processing (i.e. `Filters()`).

ns-rse, 2022-10-07:
A diagram giving and overview of the workflow process (generated using the [live preview](https://mermaid.live/edit#pako:eNqNkU1vwjAMhv9KlFMnQXfapYchoHzsMGkSu7Wo8lq3ZG0-lKQaiPLf57YcgBO55LXz-JXtnHmuC-QRryyYA_uOU8XozJOlVqWo9mw6fe9K0SDDo0flhFYdWwTG6hydy1wO6mUsWfQoWwaNhuI2vxzycZA9PsSDd-iM7NgqMafd12e4sG2Ndn8H_Jp6WmqbY8fWCZSStAQfDnYFeLinD29PYeLn7ylO-wPajm2SuRpkaCCvocIrtRmG2wbKhKoAa-HEXpkRR2wyrzMl-3kboSrK9ktUIJHkbDZj1yWsRoMxWN8G2yH4CLw22nnwLiQHj9aF6_EmBz7hEql9UdAPnvu6lFOXElMekaSO6pSn6kIctF7vTirnkbctTnhraFiMBdDHSx6V0Di8_AN-W6eD)).

```mermaid
graph TD
    A[Config] -->|file extension| B(process_scan)
    B --> C(load_scan)
    C --> D(_load_scan)
    D -->|.spm| E[pySPM.Bruker]
    D -->|.jpk-force| F[afmformat.load_data]
    D -->|.h5| F[afmformat.load_data]
    D -->|.ibw| F[afmformat.load_data]
    D -->|.other| G[Another.package]
    G --> H(np.ndarray / pixel_to_nm_scaling / filename / ??? )
    E --> H
    F --> H
    H --> I(topostats.filters.Filters)

```

ns-rse, 2022-10-12:
#306 switches from `io.load_scan()` function to `io.LoadScan()` class and sets the foundations of a factory design pattern to support additional formats. Tests were updated to work with this.

MaxGamill-Sheffield, 2024-11-13:
All implemented now, and put into AFMReader thanks @SylviaWhittle!

closed 2024-11-13 by MaxGamill-Sheffield