AFM-SPM/TopoStats #145: Parity between gwyddion TopoStats grain statistics and `dev`

SylviaWhittle, 2022-05-17:
**Is your feature request related to a problem? Please describe.**
`dev` does not produce grain statistics that are in parity with the old gwyddion TopoStats grain statistics. 

**Describe the solution you'd like**
Parity needs to be made between them before continuing with development

A summary of the statistic discrepancies:
[image]

bobturneruk, 2022-05-17:
@ns-rse can you look at this one, please? Hopefully with a shorter development cycle and smaller PRs in future user / acceptance testing can take place against `dev` and avoid this kind of thing. Kind of to be expected at this stage, though. Good spot @SylviaWhittle!

SylviaWhittle, 2022-05-17:
@bobturneruk I've made the necessary changes, just adding the issue for documentation purposes. Making a PR now.

bobturneruk, 2022-05-17:
Splendid! Looking forward to it.

SylviaWhittle, 2022-05-19:
Completed as of #147

PR #147 (2022-05-19): Sylvia whittle/145 dev dev remove gwyddion parity
Resolves #145 
Resolves #143 
Resolves #142 

This makes the necessary changes to achieve parity between `dev` and `dev-remove-gwyddion`, and thus by extension, `master`. I believe this change means that the output statistics of `grainstats.py` are accurate and can be considered usable. 

Here is evidence of the parity:

Before the changes, as shown in #145:
[image]

After the changes:
[image]
closed 2022-05-19 by SylviaWhittle