AFM-SPM/TopoStats #153: TopoStats 2.0 : Change SPM data channel to 'Height Sensor' by default

SylviaWhittle, 2022-05-19:
**Is your feature request related to a problem? Please describe.**
Previously the default channel was 'Height', this seems to give slightly inferior results when compared to those using 'Height Sensor'.

**Describe the solution you'd like**
Replace 'Height' with 'Height Sensor' in the example.yaml config file.

alicepyne, 2022-05-19:
Could we have a backup of choosing height sensor (or Z sensor) where
available and height where not to avoid not processing files from open loop
(no sensor) systems?

SylviaWhittle, 2022-05-19:
> Could we have a backup of choosing height sensor (or Z sensor) where available and height where not to avoid not processing files from open loop (no sensor) systems?
> -- Dr Alice Pyne Senior Lecturer in Soft Matter & Polymers Department of Materials Science and Engineering University of Sheffield Email: ***@***.*** Website: [www.pyne-lab.uk](http://www.pyne-lab.uk)

Certainly! Thanks for the feedback!

ns-rse, 2022-06-17:
My tuppence.

The "default" be set in the `config.yaml` that someone uses. We can provide templates with the default as `Height Sensor`. To address the issue of falling back to `Height` on images from open loop sensor you could adapt the `Filters.extract_channel()` method so that it tries to extract the user specified channel, if that fails with a particular exception then you try extracting the `Height` channel.

Currently we have
```python
    def extract_channel(self):
        """Extract the channel"""
        try:
            self.images["extracted_channel"] = self.images["scan_raw"].get_channel(self.channel)
            LOGGER.info(f"[{self.filename}] : Extracted channel {self.channel}")
        except Exception as exception:
            LOGGER.error(f"[{self.filename}] : {exception}")
```

Instead we would have the following, because [pySPM.Bruker.get_channel()](https://github.com/scholi/pySPM/blob/master/pySPM/Bruker.py) returns a broad and fairly generic `Exception` we can not be any more specific (it might be useful to have a `ChannelException` as I've attempted to define here.

```python
    def extract_channel(self):
        """Extract the channel"""
        try:
            self.images["extracted_channel"] = self.images["scan_raw"].get_channel(self.channel)
            LOGGER.info(f"[{self.filename}] : Extracted channel {self.channel}")
        except Exception as exception:
            self.images["extracted_channel"] = self.images["scan_raw"].get_channel("Height")
            LOGGER.info(f"[{self.filename}] : Extracted channel {self.channel}")
        except:
            raise ChannelException(f"Neither {self.channel} nor Height channel could be extracted")
```

Aside : The `try: ... except: ...` construct is a very "Pythonic" way of doing things, its somewhat similar to using `if: ... elif: ... elif: ... else:` but is designed for handling errors, useful article is [here](https://12ft.io/proxy?q=https%3A%2F%2Frealpython.com%2Fpython-exceptions%2F).

bobturneruk, 2022-06-17:
I may be too far from the specifics to make useful suggestions here. But, would inferring best channel name from a user specified microscope type in the config be an option?

ns-rse, 2022-06-17:
If there is a specific channel associated with a given device the parameterising the channel based on the device would work.

Have a `microscope_channel.yaml`

```yaml
microscope1: Height
microscope2: Height Sensor
microscope3: ZSensor
```
This is loaded by the `Filters()` class (as a dictionary) and used to select the channel.

Then in the `config.yaml` you define which `microscope` you have with an entry such as the following that sets the microscope type...

```yaml
microscope: microscope1
channel: # User can still over-ride a channel by adding it to the config
```

The `Filters()` class would then initialise with something like the following...

```python
    def __init__(self, 
        img_path: Union[str, Path],
        microscope: str,
        channel: str = None,
        amplify_level: float = None,
        threshold_method: str = "otsu",
        output_dir: Union[str, Path] = None,
        quiet: bool = False,
    ):
    self.microscope = microscope
    self.channel = microscope_channel[self.microscope_channel] if channel is None else channel
    ...

```

I think there might be added complications if a package other than `pySPM` were required though.

ns-rse, 2022-12-01:
Closing as we now select channel automatically based on file extension if none is specified (see the `topostats.io.LoadScan()` class, however users can specify a channel in the `config.yaml` they pass in.

closed 2022-12-01 by ns-rse