AFM-SPM/TopoStats #154: TopoStats 2.0 : Add grain removal based on height

SylviaWhittle, 2022-05-19:
**Is your feature request related to a problem? Please describe.**
Removing anomalously high-up grains would be useful as it would remove artefacts such as scars and blobs that are not wanted and would mess with the scaling of the data.

**Describe the solution you'd like**
A system that detects grains that are 2 standard deviations from the mean grain height and removes them.

SylviaWhittle, 2022-09-27:
This maybe should go in filters as it could help with flattening

SylviaWhittle, 2022-09-27:
Focus on scars as opposed to blobs

bobturneruk, 2022-10-03:
See http://gwyddion.net/documentation/user-guide-en/scan-line-defects.html

ns-rse, 2022-10-03:
I don't know if this would be of use but I realised I had previously created #202 because I'd stumbled across the [Hough Transform](https://en.wikipedia.org/wiki/Hough_transform) which is a method for identifying classes of shapes, classically straight-lines, from images.

There are methods in scikit-image that implement this (see #202 for links).

alicepyne, 2022-10-11:
Would be good to have options for how many SD in the config file? In the case of including e.g. proteins which are 10x the height of DNA.

PR #390 (2023-01-13): Sylvia whittle/remove scars
Closes #202
Closes #154 

This pull request adds the long awaited scar removal feature to TopoStats. Configuration for the scar removal has been added in the config file, and good/useful defaults have been set for the parameters. 

The scar removal is done by the new `Scars` class. Much like with the `Filters` class, its instantiation takes an image and the required parameters, then one can call the `remove_scars()` class method on the instance. `remove_scars()` is the only method intended to be used externally to the class, though it is written in a way that you co [... cut]
closed 2023-01-13 by SylviaWhittle