AFM-SPM/TopoStats #158: Dev : Masks not applied properly during second round of flattening

SylviaWhittle, 2022-05-24:
**Describe the bug**
On line `173` of `filters.py`, the masked array that is returned is actually a masked array object, with the masked values not filled in. The masked values are instead, '--'. This causes problems for many `.spm` files that one may attempt to process. 

I have a hotfix in the works.

**To Reproduce**
Run topostats on the `dev` branch with most `.spm` files.

**Expected behavior**
Line `173` in `filters.py` returns an array whose masked values are filled with `np.nan`. 

**Output**
[image]

ns-rse, 2022-05-24:
@SylviaWhittle  and I have had some discussion on this and have done some investigating on this.

# Masking

First we checked to see how masking is applied to a 1D array and how `np.median()` and `np.quantile()` handle that masking...

```python
import numpy as np

x = np.ma.array(np.arange(8), mask=[0] * 4 + [1] * 4)

print(f"Masked array  : {x}")
print(f"Masked median : {np.ma.median(x)}")
print(f"Median of [0,1,2,3] : {np.quantile(np.arange(4), [0.5])}")
print(f"Median of [0,1,2,3,4,5,6,7] : {np.quantile(np.arange(8), [0.5])}")

In [126]: 
In [127]: 
In [127]: 
In [128]: 
In [128]: Masked array  : [0 1 2 3 -- -- -- --]

In [129]: Masked median : 1.5

In [130]: Median of [0,1,2,3] : [1.5]

In [131]: Median of [0,1,2,3,4,5,6,7] : [3.5]
```

The mask is correctly applied and respected by `np.median()` but *not* by `np.quantile()` and it doesn't report the quantiles requested (25th, 50th and 75th) but instead a single value...

```python
Masked array  : [0 1 2 3 -- -- -- --]

Masked median : 1.5

/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/numpy/lib/function_base.py:4650: UserWarning: Warning: 'partition' will ignore the 'mask' of the MaskedArray.
  arr.partition(
Masked quantiles: [1.75 -- --]

Median of [0,1,2,3] : [1.5]

Median of [0,1,2,3,4,5,6,7] : [3.5]
```

A more realistic example is below with a 2-D `np.ndarray` with a mask applied.

```python
RNG = np.random.default_rng(seed=3892398)
SMALL_ARRAY_SIZE = (3, 3)
THRESHOLD = 0.5

small_array = RNG.random(SMALL_ARRAY_SIZE)
mask = RNG.uniform(low=0, high=1, size=SMALL_ARRAY_SIZE) > THRESHOLD
small_array_masked = np.ma.masked_array(small_array, mask, fill_value=np.nan)

print(f"WITHOUT MASK")
print(f"small_array :\n{small_array}")
print(f"np.quantile(small_array, [0.5]) :\n{np.quantile(small_array, [0.5], axis=1).T}")
print(f"np.median(small_array, axis=1) :\n{np.median(small_array, axis=1)}")
print(f"np.nanmedian(small_array, axis=1) :\n{np.nanmedian(small_array, axis=1)}")
print(
    f"np.nanquantile(small_array, [0.25, 0.5, 0.75], axis=1).T :\n{np.nanquantile(small_array, [0.25, 0.5, 0.75], axis=1).T}"
)

print(f"WITH MASK")
print(f"small_array_masked :\n{small_array_masked}")
print(f"np.quantile(small_array_masked, [0.5]) :\n{np.quantile(small_array_masked, [0.5], axis=1).T}")
print(f"np.median(small_array_masked, axis=1) :\n{np.median(small_array_masked, axis=1)}")
print(f"np.nanmedian(small_array_masked, axis=1) :\n{np.nanmedian(small_array_masked, axis=1)}")
print(
    f"np.nanquantile(small_array_masked, [0.25, 0.5, 0.75], axis=1) :\n{np.nanquantile(small_array_masked, [0.25, 0.5, 0.75], axis=1)}"
)
```

This gives the following...

```python
WITHOUT MASK

In [114]: small_array :
[[0.55893194 0.58423965 0.54991388]
 [0.36399804 0.06721005 0.02255171]
 [0.41724214 0.33160048 0.66949514]]

In [115]: np.quantile(small_array, [0.5]) :
[[0.55893194]
 [0.06721005]
 [0.41724214]]

In [116]: np.median(small_array, axis=1) :
[0.55893194 0.06721005 0.41724214]

In [117]: np.nanmedian(small_array, axis=1) :
[0.55893194 0.06721005 0.41724214]

In [118]: np.nanquantile(small_array, [0.25, 0.5, 0.75], axis=1).T :
[[0.55442291 0.55893194 0.57158579]
 [0.04488088 0.06721005 0.21560405]
 [0.37442131 0.41724214 0.54336864]]

In [119]: 
In [119]: WITH MASK

In [120]: small_array_masked :
[[0.5589319369888012 0.5842396462969964 0.549913882332871]
 [0.3639980357139009 0.06721005487309961 --]
 [0.41724213939486854 -- --]]

In [121]: /home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/numpy/lib/function_base.py:4650: UserWarning: Warning: 'partition' will ignore the 'mask' of the MaskedArray.
  arr.partition(
np.quantile(small_array_masked, [0.5]) :
[[0.5589319369888012]
 [--]
 [--]]

In [122]: /home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/numpy/core/fromnumeric.py:758: UserWarning: Warning: 'partition' will ignore the 'mask' of the MaskedArray.
  a.partition(kth, axis=axis, kind=kind, order=order)
np.median(small_array_masked, axis=1) :
[0.5589319369888012 0.06721005487309961 --]

In [123]: np.nanmedian(small_array_masked, axis=1) :
[0.55893194 0.21560405 0.41724214]

In [124]: /home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/numpy/lib/function_base.py:4650: UserWarning: Warning: 'partition' will ignore the 'mask' of the MaskedArray.
  arr.partition(
np.nanquantile(small_array_masked, [0.25, 0.5, 0.75], axis=1) :
[[0.5544229096608361 0.04488088464123102 --]
 [0.5589319369888012 -- --]
 [0.5715857916428988 -- --]]
```

It shows a couple of things....

1.The mask is applied without the need to call the .filled() method.
2. The `UserWarning`'s shows that both `np.quantile()` and `np.median()` ignore the mask that has been applied.
3. `np.nanmedian()` does what we would like and respects the mask.
4. `np.nanquantile()` is doing something very strange that I can't work out, but note it also gives the warning about ignoring the mask.

Overall we are only using the median so we could "easily" switch to `np.nanmedian()` instead of the existing `np.quantile()`. I say "easily" because actually quite a few tests will need updating in light of this as its going to give different answers for quite a few things, also  other sections of code will need tweaking to use the median rather than pulling the median out of the arrays of quantiles.

ns-rse, 2022-06-01:
Closed by #161