AFM-SPM/TopoStats #159: Dev: "01-raw_heightmap.png" plots processed data instead of raw height data

MaxGamill-Sheffield, 2022-05-25:
**Describe the bug**
The output plot produced to show the raw heightmap (01-raw_heightmap.png) shows data that looks like it has undergone some processing already and thus is not the raw height data from the SPM file.

**To Reproduce**

- Command used to run topostats e.g. `python pygwytracing.py` / `python topostats/topotracing.py`
- Config file (original config file used)
- View output image "01-raw_heightmap.png"

**Expected behavior**
The raw AFM data before any processing should be shown in this image.

**Output**
Output image attached below.
[image]

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: bf56931 (HEAD -> dev, origin/dev) Merge pull request #155 from AFM-SPM/ns-rse/90a-refactor-classes.
 - Python version: 3.8.13.
 - Packages attached.
[packages.txt](https://github.com/AFM-SPM/TopoStats/files/8770764/packages.txt)

**Additional context**
N/A

bobturneruk, 2022-05-25:
Can you share the file, please? Has the file perhaps been edited before being used with topostats?

MaxGamill-Sheffield, 2022-05-25:
@bobturneruk I'm not sure how to share the spm files on GitHub, but the file is the mincircle.spm that was shipped with TopoStats and looks like this when loaded into gwyddion

[image]

ns-rse, 2022-06-16:
Just noticed this. The current first image I could get to plot was `pixels` as trying to plot the results of `load_scan()` stored in `self.images["scan_raw"]` (see line 97 of `filters.py`) resulted in and error...

```
AttributeError: 'Bruker' object has no attribute 'show'
```

I've gone and read the [documentation for pySPM](https://nbviewer.org/github/scholi/pySPM/blob/master/doc/pySPM%20Documentation.ipynb) and see it includes an example of plotting extracted channels so will try and use that as a basis to get the correct image out.

PR #180 (2022-06-16): 159 | Introduces correct "raw_height" plot; adds write_yaml()
Resolves #159 

Plots the `raw_height` of the Bruker image to `00-raw_height.png` resolving #159.
The reason I'd not included this previously is I was confused why the `pySPM.SPM` that the `extract_channel()` plots couldn't be plotted, its because this object has the attribute `pixels` which is what should be plotted.

That said I'm not entirely sure why this, which is the plot of `pySPM.SPM.pixels` differs from the plot of pixels as I can't see why/where the `self.images["pixels"]` is modified in comparison to the extraction and so [... cut]
closed 2022-06-16 by ns-rse