AFM-SPM/TopoStats #160: Dev: Minimum bounding box plot showing wrong bounding box

MaxGamill-Sheffield, 2022-05-26:
**Describe the bug**
The plots showing the minimum bounding box does not show the correct minimum bounding box. However, values obtained for the bounding length and width are from the correct bounding box.

**To Reproduce**

- Command used to run topostats:  `python pygwytracing.py` / `python topostats/topotracing.py`
- Config file: default config file 
- Processed file: the standard minicircle.spm file
- Observe "minimum_bbox.png" plot output for grain 11

**Expected behavior**
The plot produced for each grain labeled "minimum_bbox.png" (image attached) should look similar to the blue line and not the output orange one.
This is because the minimum bounding width is the "narrowest street the car can drive down" and as should be the smallest dimension of any box that covers the molecule. 
[image]

**Output**
The plot output for grain 11 is shown below, which does not look like the smallest bounding width.
[image]
The smallest bounding box statistics are calculated correctly and shown below.
[image]

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS
 - topostats version: bf56931 (HEAD -> dev, origin/dev) Merge pull request #155 from AFM-SPM/ns-rse/90a-refactor-classes
 - Python version: 3.8.13

SylviaWhittle, 2022-08-02:
The calculations behind finding the minimum bounding box are correct, the script that I use to plot the bounding box has a bug in it where the box that is plotted is not the minimum bounding box. I suggest that we remove the plotting since this was a feature that was solely used for development purposes.

MaxGamill-Sheffield, 2024-11-13:
we don't make these plots anymore, @SylviaWhittle sure the issue was in the plotting and not the ferets

closed 2024-11-13 by MaxGamill-Sheffield