AFM-SPM/TopoStats #188: exit code -2147483645 when running with PyCharm and few grains are identified

Jean-Du, 2022-06-28:
**Describe the bug**
When running with PyCharm, and very few grains are identified, the message 'Process finished with exit code -2147483645' is produced, and the masked images are not produced.

**To Reproduce**

Files producing the error: https://drive.google.com/drive/folders/1OgDYPccXs1EMiBNAzCkxIArurh6bfSkB
The branch 'Curvature' is used, with the main relevant difference being the line `minsize = int(minarea / dx)`  (line 235 in pygwytracing.py for master branch) is changed to `minsize = int(minarea / dx**2) ` so that minarea represents an actual area instead of a length.
The parameters used are
```
minarea = 1350e-18
maxdeviation = 100.0
mindeviation = 0.0
gaussian = 0.1e-9
thresholdingcriteria = 2.0
```
and the image output options are
```
; Turn saving trace figures on and off. 1 : on, 0 : off
saveTraceFigures_option = 0
; Turn saving cropped files on and off. 1 : on, 0 : off
saveCroppedFiles_option = 1
; Save file scale option: 0 : none, 1 : ruler, 2 : inset scale bar
savefilesScale_option = 2
; Save file name for output colour scale eg: Nanoscope.txt
savefile_zscalecolour = Gwyddion.net
; Turn on and off the z bar. 1 : on, 0 : off
zbar_option = 1
```

**Expected behavior**
Expected 'Process finished with exit code 0' and output files in the 'processed' folder.

**Output**

The message 'Process finished with exit code -2147483645' is produced while analysing one of the files (https://drive.google.com/file/d/1QO2P6BE_WSZxXLHcR_F_erOOqsDOWuln/view?usp=sharing)

If applicable please include the output error, this can be a copy and paste of the output (preferable) or a screenshot.
If applicable, add screenshots to help explain your problem.
Output files can be attached to this bug report.

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: Windows 10, version 21H2
 - topostats version: 90eb901
 - Python version: 2.7
 - Optionally, your installed packages: best way to get this is to paste the results of typing `pip freeze`.

**Additional context**
Add any other context about the problem here.

bobturneruk, 2022-06-28:
When the file in question is the **only** file in the folder being analysed the error does not occur.

bobturneruk, 2022-06-28:
If we think it might be a memory error, this may help https://github.com/bloomberg/memray

bobturneruk, 2022-06-28:
Also from @Jean-Du 👍 

>When I tried to run this through the command line (instead of using PyCharm), I got  GLib-ERROR **: ../../glib/gmem.c:100: failed to allocate 8388608 bytes

So maybe Gwyddion's C code?

ns-rse, 2022-06-28:
An aside and no idea how much RAM is on the machines where this occurs but that is ~8Mb that there is a memory allocation error on which is surprising as most machines have > 8GB these days. Even with multiple files at ~32.1Mb and several at ~8Mb there should still be enough RAM. Maybe, since the sizes are ~8Mb this is related to cropped images in some manner?

I won't be able to test this on the new pipeline for a few days at least as I've other aspects to focus on, but the steps should be "relatively easy" to try out, something like...

```
cd <wherever/topostats/repository/exists>
git checkout dev
git pull
conda create --name topostats-test-mem-fail python=3.9
conda activate topostats-test-mem-fail
pip install -e .
run_topostats --config config/example.yaml --base_dir <path/to/where/the/problem/files/are>
```

Jean-Du, 2022-06-28:
Further testing seems to show that this related to how many files it has to process at the same time, and not a specific file.

bobturneruk, 2022-06-28:
I wonder if memory can be explicitly purged between files.

Jean-Du, 2022-12-06:
@ns-rse Since this one is only relevant to old TopoStats, should we close it?

SylviaWhittle, 2023-02-16:
Only affecting old Topo, closing 👍

closed 2023-02-16 by SylviaWhittle