AFM-SPM/TopoStats #201: Dev and 165: Error 17 when the output directory specified in the config file doesn't exist

Jean-Du, 2022-07-11:
**Describe the bug**
When the output directory specified in the config file doesn't already exist, this directory is created, but the following error occurs:
`FileExistsError: [Errno 17] Cannot create a file when that file already exists:`, followed by the path to the output directory.
Running it again works because this time the directory already exists.

**To Reproduce**

- Command used to run topostats: `run_topostats --config config/example.yaml`
- Config file (you can attach this or paste it below)

**Expected behavior**
The output directory is created and the code runs normally

**Output**

`Traceback (most recent call last):
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\run_topostats.py", line 426, in <module>
    main()
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\run_topostats.py", line 353, in main
    config = read_yaml(args.config_file)
  File "c:\users\mingxue\pycharmprojects\topostats\topostats\io.py", line 28, in read_yaml
    with Path(filename).open() as f:
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\pathlib.py", line 1252, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\pathlib.py", line 1120, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: 'config\\example.yaml'`

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: Windows 10, version 21H2
 - topostats version: 4af3876
 - Python version: 3.10
 - Optionally, your installed packages: best way to get this is to paste the results of typing `pip freeze`.

**Additional context**
Add any other context about the problem here.

Jean-Du, 2022-07-11:
Update: This only occurs when the output directory is NOT under TopoStats

SylviaWhittle, 2022-07-28:
It has been suggested by the experimentalists, that we should support and use absolute paths by default. I think we should have a toggle in the config files `absolute_path` for example.

bobturneruk, 2022-07-28:
Python can tell if a path is absolute without the user explicitly stating this https://docs.python.org/3/library/os.path.html#os.path.isabs

MaxGamill-Sheffield, 2022-07-28:
@SylviaWhittle @bobturneruk I've just tested the abs path (on mac at least) on [my branch](https://github.com/AFM-SPM/TopoStats/tree/maxgamill-sheffield/224-truncate-output-and-img-names) and it seems like abs paths are working.

bobturneruk, 2022-07-28:
Sounds good. Is this potential bug covered by a unit test?

MaxGamill-Sheffield, 2022-07-28:
@bobturneruk I believe there is a test which checks to see if the output dir exists and passes. But I think @Jean-Du means this occurs only when a full path is specified which I don't think we can test for because of varying test file systems

SylviaWhittle, 2023-02-16:
Cannot reproduce via instructions. I think this may be an OS-specific issue? I searched online and found mentions of this in relation to already open, or corrupted files.

ns-rse, 2023-02-16:
This shouldn't be happening at all since all of the `.mkdir()` calls include the option `exist_ok=True`.

There are 10 instances across the files (I purposefully exclude the `topostats/tracing/` directory since that is a work in progress and the refactoring doesn't do any writing of files).

Count how many matches and then how many of those matches include `exist_ok` and its the same number.

```bash
❱ grep mkdir topostats/*.py   
topostats/grainstats.py:            base_output_dir.mkdir(parents=True, exist_ok=True)
topostats/grainstats.py:                path.mkdir(parents=True, exist_ok=True)
topostats/io.py:                out_path_processed.mkdir(parents=True, exist_ok=True)
topostats/plotting.py:        self.output_dir.mkdir(parents=True, exist_ok=True)
topostats/processing.py:    core_out_path.mkdir(parents=True, exist_ok=True)
topostats/processing.py:    filter_out_path.mkdir(exist_ok=True, parents=True)
topostats/processing.py:    Path.mkdir(grain_out_path / "upper", parents=True, exist_ok=True)
topostats/processing.py:    Path.mkdir(grain_out_path / "lower", parents=True, exist_ok=True)
topostats/run_topostats.py:    config["output_dir"].mkdir(parents=True, exist_ok=True)
topostats/run_topostats.py:            summary_config["output_dir"].mkdir(parents=True, exist_ok=True)
(topostats) ❱ grep mkdir topostats/*.py  | wc -l
10
(topostats) ❱ grep mkdir topostats/*.py  | grep exist_ok | wc -l
10
```

I suspect this may now be solved but it would be useful if you could check please @Jean-Du, if we can close this out its one more bug out of the path to a new release.

Jean-Du, 2023-02-17:
Hi @ns-rse , I just checked with the latest main branch (commit 06dabbe8106c26100957cd1b346b8218ce988e18) and unfortunately I still get this error

Jean-Du, 2023-02-17:
This is the error message I get:
```
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\processing.py", line 78, in process_scan
    core_out_path.mkdir(parents=True, exist_ok=True)
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\pathlib.py", line 1323, in mkdir
    self._accessor.mkdir(self, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'output_testdir\\processed'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Mingxue\anaconda3\envs\topostats\Scripts\run_topostats.exe\__main__.py", line 7, in <module>
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\run_topostats.py", line 213, in main
    for img, result in pool.imap_unordered(
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\pool.py", line 870, in next
    raise value
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "C:\Users\Mingxue\PycharmProjects\TopoStats\topostats\processing.py", line 78, in process_scan
    core_out_path.mkdir(parents=True, exist_ok=True)
  File "C:\Users\Mingxue\anaconda3\envs\TopoStats\lib\pathlib.py", line 1323, in mkdir
    self._accessor.mkdir(self, mode)
FileExistsError: [Errno 17] Cannot create a file when that file already exists: 'output_testdir\\processed'
```
And this is the config file I am using:
```
# Sample configuration file auto-generated : 2023-02-13 16:57:07
base_dir: ./
output_dir: output_testdir
warnings: ignore
cores: 2
quiet: false
file_ext: .spm
loading:
  channel: Height
filter:
  run: true
  threshold_method: std_dev
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  gaussian_size: 1.0121397464510862
  gaussian_mode: nearest
  remove_scars:
    run: true
    removal_iterations: 2
    threshold_low: 0.25
    threshold_high: 0.666
    max_scar_width: 4
    min_scar_length: 16
grains:
  run: true
  threshold_method: std_dev
  otsu_threshold_multiplier: 1.0
  threshold_std_dev:
    lower: 10.0
    upper: 1.0
  threshold_absolute:
    lower: -1.0
    upper: 1.0
  direction: upper
  smallest_grain_size_nm2: 1
  absolute_area_threshold:
    upper:
    - 300
    - 3000
    lower:
    - 
    - 
grainstats:
  run: true
  edge_detection_method: binary_erosion
  cropped_size: 40.0
dnatracing:
  run: true
plotting:
  run: true
  save_format: png
  image_set: core
  zrange:
  - 
  - 
  colorbar: true
  axes: true
  cmap: nanoscope
  mask_cmap: blu
  histogram_log_axis: false
  histogram_bins: 200
summary_stats:
  run: true
  config:
  ```

ns-rse, 2023-02-17:
Thanks @Jean-Du I've just given the above config a whirl against the current `main` branch and I tried three ways...

1. Without any `output_testdir` so `run_topostats` creates it.
2. With a pre-existing `output_testdir` so that when `run_topostats` tries to create it it shouldn't complain because we use the option `exist_ok=True` in all `mkdir()` calls.
3. Running it again after a successful run so that _all_ created directory structures were in place.

In all cases it runs to completion without any problem.

I'm wondering whether this is perhaps being run on a network mounted drive perhaps as that may cause some problems if perhaps the `run_topostats` process doesn't have the correct permissions to create directories. I've a very vague recollection I think from a past issue mentioning that you, quite rightly, keep your data on network drives which is why I'm wondering if this could be the underlying cause.

Its strange because you have presumably been able to use `run_topostats` from the `main` branch successfully in other instances, so there must be something unique about this particular run that is throwing this error. :thinking:

Jean-Du, 2023-02-17:
@ns-rse Thanks a lot for your response! I tried to place my data in a local folder instead of a network drive, and this error has disappeared. 
Another thing worth mentioning is that, I only get this error in your test case 1), i.e., when the output directory didn't exist and topostats had to create it

Jean-Du, 2023-02-20:
I forgot to add that I am working in a directory that I created and should have all the permissions. It's not in a shared drive, and I am the only person with access to it.

ns-rse, 2023-02-20:
> @ns-rse Have you tried running it in a directory with multiple files? I think what happened in my case is that, it processes the first file fine, then when it tried to process the second file and create something new, it finds that it has already been created when processing the first file

No, because the original report only included one file and so I used that. And I still don't understand why it happens because all of the `mkdir(..., exist_ok=True)` i.e. include the option that allows the directory to already exist and shouldn't therefore error.

> I forgot to add that I am working in a directory that I created and should have all the permissions. It's not in a shared drive, and I am the only person with access to it.

As much as I'd love to get to the bottom of this I feel my time would be spent more productively on refactoring code (DNA tracing) which is already being built upon by yourself and @MaxGamill-Sheffield and rounding out notebooks otherwise it won't get done.

There is a simple work around which places a small burden on the user to copy files to a local directory where the issue has not (yet?) been observed.

Jean-Du, 2023-02-20:
@ns-rse Hi Neil, you're absolutely right that refactoring dnatracing should be prioritised over this. As for workarounds, there is no need to copy it to a local directory; I mentioned in the original report that when this happens, running it for a second time resolves the problem, since this time it changed from scenario 1)  to scenario 2)

ns-rse, 2023-02-20:
Very confusing, why would an `FileExistsError` occur on the first instance...when it doesn't exist? Besides which any `mkdir(..., exist_ok=True)` should still work even if the directory doesn't exist in the first instance (and they all include the `parents=True` option too so if there is an intermediary directory that is expected but doesn't exist it too should be created).

It shouldn't matter that multiple files are being processed either.

Directory permissions can be a nightmare, the University is using a system called CIFS (Common Internet File System). How this is mounted depends on the OS, Windows likely uses SMB (Sever Message Block). Directories and files get mounted with three sets of read/write/execute permissions.

To investigate further I would have to sit down and try it all out myself and figure out what is going on.

PR #473 (2023-02-27): Adding FAQ and Glossary; tidying configuration
Closes #201

Adding documentation in the absence of being able to resolve network drive errors.
closed 2023-02-27 by ns-rse