AFM-SPM/TopoStats #224: Reorder output file structure

MaxGamill-Sheffield, 2022-07-19:
Just spoke to Alice, Eddie and Libby and there are concerns about the output files. Mainly to increase the speed of finding what you need and the labelling of images. See below:

When dealing with multiple files, and multiple folders of files, the outputs should:
1. Carry the name of the parent folder with them.
2. Produce a '_processed' image and '_processed_masked' image for every .spm, and a grainstats.csv of all .spm's within that folder. Additionally a directory with cropped images (as an option in the config)
3. File names should be contained within the produced images.

An example structure is shown below:
```
Input:
-- <data> --- folder1 ---- one.spm
		      ---- two.spm
		      ---- three.spm
	
	  --- folder2 ---- four.spm
		      ---- five.spm

Output:
-- <output> --- folder1_processed ---- one_processed.png (image 08_5)
		      		  ---- one_processed_masked.png (image 08_5 + 13)
				  ---- two_processed.png 
		      		  ---- two_processed_masked.png 
				  ---- three_processed.png 
		      		  ---- three_processed_masked.png
				  ---- grainstats.csv (only containing one, two and three)
				  ---- cropped...
	
	    --- folder2_processed ---- four_processed.png
				  ---- four_processed_masked.png
			   	  ---- five_processed.png
			   	  ---- five_processed_masked.png
				  ---- grainstats.csv
				  ---- cropped...

```

ns-rse, 2022-07-19:
One thing I found particularly confusing when first getting involved was the order in which plots should be considered.

I appreciate that many are redundant and not required (based on the above example) but it is why I introduced the numerical prefixes to output images.

Also `grainstats.csv` will always be produced but there is also output from `dnatracing.py` which is combined. I set it up so that they were combined, should these be included in the output of the file `grainstats.csv`?

I also aggregated output for each image (i.e. grain statistics and dna tracing statistics) across all images and output to CSV at the higher level `<output>` directory. Is this useful and desirable? It saves users having to combine the files and has an indicator of the image and so is highly conducive to Pandas `groupby()` methods when it comes to subsequent summarisation and plotting (the later code base could do with a large refactoring to take advantage of the built-in plotting methods that Pandas has).

/cc @alicepyne @derollins (not sure of Libby's GitHub account to include it).

Jean-Du, 2022-07-19:
@ns-rse Aggregating the output for each image and putting them together in one file is useful and desriable. With old TopoStats, Plotting.py was used to plot the statistical distribution for data from all files combined. However, since the scrit works by having the user specify the path to the csv (or json in old TopoStats) file, Plotting.py is equally compatible with plotting the outputs from an individual file, should that be preferred by the user.
In the future, the specification of the path should be moved from Plotting.py to the config file.

Jean-Du, 2022-07-19:
In my opinion the main compatibility issue regarding Plotting.py and new TopoStats is that, Plotting.py was designed to be highly customisable, with the user deciding which plots of which variables they need, by adding lines of code at the end of the script, such as `plotkde(df, 'grain_mean', grouparg = 'directory', xmin=0, xmax=4, nm=True)`, meaning they want a kde plot of the variable 'grain_mean', plotted over the range between 0 and 4, grouped by directory and using nm as the unit (old TopoStats output defaulted to m, Plotting.py gave the user the option to plot in either nm or m). 

However, new TopoStats was designed so that the user doesn't have to do anything with the code and only needs to run a comand through the command line. The amount of customisation for Plotting.py might be difficult to achieve in a config file, especially since the user would want to plot different number of plots, and of different things, for each run. I don't know what's the best way to resolve this and would like to hear from @ns-rse and @SylviaWhittle

MaxGamill-Sheffield, 2022-07-20:
@ns-rse In regards to all the numbered plots, I think they could still be useful for developers when trying to troubleshoot why the image tests may not be working due to their modified code. Perhaps we could add a parameter for this somewhere to truncate the outputs or not, what do you think?

ns-rse, 2022-07-20:
@MaxGamill-Sheffield Yes, something in the configuration like...

```
plotting:
  images: all # Options : all, core
```

...which controls which images are plotted would be one way of controlling that, with `all` resulting in all plots and `core` just the ones commonly of interest.

ns-rse, 2022-08-05:
Looks like this is complete, can it be closed?

closed 2022-08-05 by MaxGamill-Sheffield