AFM-SPM/TopoStats #227: Refactor Plotting.py

ns-rse, 2022-07-19:
`topostats/Plotting.py` would benefit from refactoring. Ideally it should be a class with callable methods and would benefit from...

* Use of `pathlib` for filename and directory operations.
* Correct use of logging rather than print statements.

Outstanding issues that mention Plotting...

* #29
* #56 
* #57 
* #87 
* #138 
* #205 
* #224 
* See also comment from @Jean-Du [here](https://github.com/AFM-SPM/TopoStats/issues/224#issuecomment-1189390563)

Something to consider is to move away from running `Plotting` as a script that one tries to configure (although there should be sensible defaults to produce starting points) and instead allow users to run Jupyter Notebooks to produce the exact plots they want, as this would allow users to tweak things as they desire. No template Notebooks are required, but good documentation on how to use the Class would be essential.

bobturneruk, 2022-07-21:
With plotting code in packages, in general... there are a few ways to go:

- Provide no plotting functionality at all and rely on users to code their own plots e.g. in a Jupyter notebook.
- Provide a core set of plotting functions for commonly used charts and rely on the user to write custom code for more specific plots. These could be called from notebooks.
- Provide comprehensive plotting functionality and add new charts in response to user requests.

Encouraging more use of notebooks guides people towards better practises e.g. away from Excel but creates a potential barrier to entry. This can be partially overcome by offering tutorials, example notebooks and training, but may be insurmountable for some.

Providing comprehensive and responsive plotting within the package typically creates a lot of boilerplate code, much of which may be very infrequently executed. e.g. https://github.com/SheffieldML/GPy/tree/devel/GPy/plotting

I suggest workshopping some notebook plotting to potential users to gauge interest / horror.

ns-rse, 2022-07-22:
Make plotting optional across a run optional in terms of both the scans, the cropped grains and the KDE plots.

bobturneruk, 2022-10-11:
Some work here https://github.com/AFM-SPM/TopoStats/tree/JeanDu/227-RefactorPlotting

ns-rse, 2022-10-11:
Met with @Jean-Du last week and she's done a great job getting this in shape on that branch. I intend to refactor this by 

* [ ] linting the code with `black`.
* [ ] linting the code with `pylint`.
* [ ] refactor to a class.
* [ ] streamline and incorporate the configuration and plotting into `run_topostats` entry point.
* [ ] provide a separate entry point that works with a stand-alone plotting configuration.
* [ ] Tests, lots of tests!
* [ ] probably more things I'm yet to think of.

May tackle this before the long-standing refactoring of DNA Tracing as its a bit more straight-forward.

Jean-Du, 2023-01-31:
I had a look at @ns-rse 's plotting branch and here are my thoughts:
1. The violin plot at the moment compares the value range of different parameters, which isn't very useful. It would be very useful to have violin plots on a single parameter, grouped by which directory the file was in, i.e. the value of the 'basename' column in the statistics csv. In this way we will be able to compare, eg the area distribution of molecules under two different experimental conditions. 
2. The config file only lets you set x range once, but different parameters will need different x range. Ideally this should be something that's configurable for each individual plot?
3. It would be very useful to allow the user to filter data based on values of certain column: eg plot only if area is greater than a certain value and molecule is linear
4. With regards to the grouping mentioned in point 1, would also be useful to allow the user to enter multiple csv files and compare them in the violin plot (or grouped histograms)

ns-rse, 2023-02-03:
Thanks for the feedback @Jean-Du 

Finally got round to updating the branch to `main` and starting to look through this, the following are first thoughts...

> 1. The violin plot at the moment compares the value range of different parameters, which isn't very useful. It would be very useful to have violin plots on a single parameter, grouped by which [...]

Yep, can do this, pretty easy too, its might even have been what I intended to do. A lot of the code was half-finished.

> 2. The config file only lets you set x range once, but different parameters will need different x range. Ideally this should be something that's configurable for each individual plot?

Making this highly configurable isn't going to work with the idea of producing a default set of plots, because you won't know what the range of x-range for each set until its processed. Part of my thoughts on this was not just a stand alone plotting but an additional step in the pipeline of image processing. Those values are going to change on each set of images and so it becomes an iterative process of running a script, its far more sensible to switch to Notebooks load the CSV and tweak the returned plots to the desired fit.

One thing I wasn't happy and hadn't figured out was not being able extend the x-range across to accomodate the KDE lines where they reached the asymptote of zero.

> 3. It would be very useful to allow the user to filter data based on values of certain column: eg plot only if area is greater than a certain value and molecule is linear

Filtering on molecule type could easily be acomodated but the filtering on area suffers from the same issue I describe above, you have to do the processing once before you can then know what value you want to filter on. Again loading the data in a Notebook and tweaking the returned plots.

> 4. With regards to the grouping mentioned in point 1, would also be useful to allow the user to enter multiple csv files and compare them in the violin plot (or grouped histograms)

I think that could be accommodated, although given the desire to have an `all_statistics.csv` and a `folder_grainstats.csv` output to each directory it has potential to result in duplicated observations (since the later are subsets of the former) and so if the code was written as "_Search this directory for files with `.csv`_" then they will all get picked up. 

This wouldn't be an issue if running as part of the pipeline of processing since the refactoring I've done allows either a `.csv` to be loaded or a `pd.DataFrame` to be passed directly which would be the natural way of doing it straight after processing a bunch of files (its unnecessary overhead and slower to read a file which is already in memory).

Jean-Du, 2023-02-06:
> Part of my thoughts on this was not just a stand alone plotting but an additional step in the pipeline of image processing. 

My thoughts are that it would function as both. A default set of plots can be produced as part of the pipeline, without user configurated x range. The user can also run the plotting script on its own and adjust the parameters, which might be more useful than the default plots, especially since the variables that need to be plotted (eg area, height, maximum bounding distance) are highly dependent on the project. I do agree that Jupyter Notebook seems to be a better format, but as the current format has the advantages of requiring less learning and automatically updating the figure in the same location (which is useful in combination with Adobe Illustrator), would it be possible to have both? We might want to discuss this in the next TopoStats meeting when @alicepyne is around.

> Filtering on molecule type could easily be acomodated but the filtering on area suffers from the same issue I describe above, you have to do the processing once before you can then know what value you want to filter on.

I would say that this isn't necessarily the case. For example, in my bending angle project, I know before I process the data that I am only interested in molecules longer than a certain length, so I believe this idea is still compatible with the set of default plots.

ns-rse, 2023-02-06:
Hi @Jean-Du 

Yes what I'm writing/working on will do both, tag onto the end of `run_topostats` and run as stand-alone. 

One solution I thought of and almost done implementing is to save the generated images as a "pickle" (Python's term for saved objects) and people can then load those and tweak the plots as required since they are ultimately just Matplotlib objects.

I hope to have a Pull Request out later today or tomorrow.

Neil

Jean-Du, 2023-02-06:
Hi @ns-rse ,
That sounds very useful! Thanks a lot for working on this and I am looking forward to seeing the pull request.

PR #455 (2023-02-10): Refactoring plotting (and more)
Closes #227
Closes #451 
Closes #452

## Refactoring plotting.py

This is the initial refactoring of `plotting.py`. Because of the vast scope for variation that exists in generating plots a deliberate attempt to keep a sensible and minimal set of plots has been taken whilst refactoring the code to be as succinct as possible. Originally there was a mixture of Matplotlib and Seaborn functions, to maintain simplicity the refactoring uses only Seaborn.

* Histograms and Kernel Density plots can be generated (if both are requested then the KDE is overlaid).
* Violin  [... cut]
commit 256420a5 (2023-02-10): 227 | Refactoring plotting (and more)

Closes #227 #451 #452

Refactoring plotting.py

This is the initial refactoring of `plotting.py`. Because of the vast scope for variation that exists in generating
plots a deliberate attempt to keep a sensible and minimal set of plots has been taken whilst refactoring the code to be
as succinct as possible. Originally there was a mixture of Matplotlib and Seaborn functions, to maintain simplicity the
refactoring uses only Seaborn.

* Histograms and Kernel Density plots can be generated (if both are requested then the KDE is overlaid).
* Violin plots can b [... cut]
commit f107ef70 (2023-02-10): 227 | Refactoring plotting (and more)

Closes #227 #451 #452

Refactoring plotting.py

This is the initial refactoring of `plotting.py`. Because of the vast scope for variation that exists in generating
plots a deliberate attempt to keep a sensible and minimal set of plots has been taken whilst refactoring the code to be
as succinct as possible. Originally there was a mixture of Matplotlib and Seaborn functions, to maintain simplicity the
refactoring uses only Seaborn.

* Histograms and Kernel Density plots can be generated (if both are requested then the KDE is overlaid).
* Violin plots can b [... cut]
closed 2023-02-10 by ns-rse