AFM-SPM/TopoStats #232: Move gaussian filter from Grains() to Filters()

ns-rse, 2022-07-22:
Currently the Gaussian filtering of the image is performed as one of the first steps of processing in the `Grains()` class. This has the undesired effect that the base image required is not generated if no grains are found.

Moving the application of Gaussian filter to the `Filters()` class will remedy this and mean there is always an image available even if no grains are found (or that step fails).

This will also require reconfiguring the plotting of images in `process_scan()` (defined in `topostats/run_topostats.py`).

ns-rse, 2022-08-05:
I'm looking at doing this but have a question.

The thresholds are currently calculated based on the zero-averaged-background image see `find_grains()` lines 264-270.

If we switch to producing the Gaussian filtered image in `Filters()` should we also calculate the threshold in `Filters()` and pass that to the `Grains()` class?

Or should this threshold actually be being calculated on the Gaussian filtered image?

An alternative, which results in code duplication, would be to produce a Gaussian filtered image in `Filters()` and maintain the current processing of `Grains()` using the `zero-averaged-background`.

I'm not a fan of duplicating code, but one of the reasons for making the move is so that if no grains are found users will still have a Gaussian Filtered image available.

MaxGamill-Sheffield, 2022-08-05:
> I'm looking at doing this but have a question.
>
> The thresholds are currently calculated based on the zero-averaged-background image see `find_grains()` lines 264-270.
>
> If we switch to producing the [...]

Yep you've got it - the threshold is being calculated from the Gaussian filtered image line 278.

So line 264 is where the `self.thresholds` dict is made but nothing from that dict is used in `self.gaussian_filter()` or `self.z_thresholding()`, nor is the `direction`. Hence we should be able to move these and the associated grain params in the config file such as `gaussian_size/_mode` to the Filters module.

MaxGamill-Sheffield, 2022-08-18:
Also look to see if the config param `background` averages the background and should possibly be moved to the filters section?

MaxGamill-Sheffield, 2022-08-30:
Working on a branch for this, but I wanted to explain changes in the image results and tests:

Previously, topostats used `zero_averaged_background` as the image passed into `Grains()`. This meant that all the thresholds were taken from this image instead of the gaussian filtered image (as obtaining thresholds was done at the start of `Grains.find_grains`).
As the passed image is now `gaussian_filtered`, this now generates a different threshold value and thus the produced images and statistics are slightly different.

MaxGamill-Sheffield, 2022-08-31:
The above has also meant that now 6 "lower" grains are identified opposed to the earlier 1.

MaxGamill-Sheffield, 2022-09-08:
Closed by #280

closed 2022-09-08 by MaxGamill-Sheffield