AFM-SPM/TopoStats #236: readthedocs documentation is currently failing to build from the `dev` branch.

bobturneruk, 2022-07-27:
**Describe the bug**
readthedocs documentation is currently failing to build from the `dev` branch.

https://readthedocs.org/projects/topostats/builds/17510515/

This needs investigation.

ns-rse, 2022-07-28:
Looking at the failure point and going back through the logs its installing...

```
Collecting sphinx-markdown-tables
  Downloading sphinx_markdown_tables-0.0.15-py3-none-any.whl (28 kB)
```

As I discovered on another package there has been a breaking change [Markdown 3.4 causes TypeError due to breaking changes](https://github.com/ryanfox/sphinx-markdown-tables/issues/36#).

And as expected the logs show that the `markdown >= 3.0.1` results in `markdown-3.4.1` being installed, but not the newer version `sphinx-markdown-tables-0.0.16` which released four days ago and included the [fix](https://github.com/ryanfox/sphinx-markdown-tables/pull/37) addressed this issue.

In turn however the requirement for the `markdown` package itself needs upgrading to `markdown > 3.4` (see [Latest version should declare dependency on markdown&gt;=3.4 · Issue #38 · ryanfox/sphinx-markdown-tables](https://github.com/ryanfox/sphinx-markdown-tables/issues/38)).

Not familiar with BuildTheDocs, but could perhaps consider building the documentation and deploying to GitHub Pages as an alternative (or as well)?

bobturneruk, 2022-07-28:
I'd be in favour of using gh pages so we don't have another site / service to manage.

ns-rse, 2022-08-04:
[ReadTheDocs](https://readthedocs.org/projects/topostats/builds/17622395/) build is working again. I'm going to leave that in place and setup GitHub pages in parallel for now. ReadTheDocs can be disabled once these are up and running, or just leave in place, they are mirrors of each other anyway since they are built from the same source.

ns-rse, 2022-08-15:
Closed with #251 documentation now hosted on [ReadTheDocs](https://topostats.readthedocs.io/en/dev/) _and_ [GitHub Pages](https://afm-spm.github.io/TopoStats/).

PR #251 (2022-08-04): 236 | Adding workflow to build docs on GitHub Pages
Resolves #236 

Adds a workflow to build the website and host it on GitHub Pages.
closed 2022-08-15 by ns-rse