AFM-SPM/TopoStats #240: Extracted real length of images sometimes wrong

MaxGamill-Sheffield, 2022-07-29:
**Describe the bug**
When processing certain images, the `Filters.extract_pixel_to_nm_scaling` function using the `.get_extent()` attribute returns a value of 1 instead of 1um. Thus the produced images do not undergo the correct amount of Gaussian blurring to remove bits of noise within the image.

**To Reproduce**

- Request the file (available on slack but not here due to sensitive information)
- Run Topostats on that file using default parameters and observe the log output for `Pixels to nm scaling : 0.0009765625`
- This is significantly shorter for what is expected (1000nm / 1024px = 0.976...)
- And thus small bits of white noise are not removed at the gaussian blurring stage 

**Expected behavior**
The px to nm scaling value should be close to 1 and the correct value obtained from the `get_extent()` call.

**Output**
Output explained above. a print statement at line 124 with `print("Length: ", self.images["extracted_channel"].get_extent()[1])` will show the value obtained for the real length.

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: e.g. MacOS 12.4
 - topostats version: 18cf37c (HEAD -> dev, origin/dev) Merge pull request #231 from AFM-SPM/maxgamill-sheffield/224-output-file-dir-structure
 - Python version: 3.10.5

**Additional context**
I wonder if the value resets to 1 every scale of 1000, i.e. if the scan is 50x50nm, a value of 50 is obtained, and if 50x50um, a value of 50 is still obtained?

MaxGamill-Sheffield, 2022-08-09:
Looking at the pySPM source code, the function `get_extent()` returns no units.

Using the following, one can obtain the units of the scan:
```
data = pySPM.Bruker(<fpath>).get_channel('Height')
data.size['real']['unit']
```

This returns:
- 'nm' for nanometres
- 'um' for micrometers
- (I don't have any larger or smaller data for the rest)

As this seems like it is encoded as part of reading the file (`'unit': scan_size[2].decode(encoding)`), I don't think there is a way to extract the actual number and maybe the solution is to create a conversion table?

alicepyne, 2022-08-09:
A conversion table makes sense to me! Good find

closed 2022-08-16 by MaxGamill-Sheffield