AFM-SPM/TopoStats #241: Make DNA Tracing Optional

MaxGamill-Sheffield, 2022-08-02:
**Is your feature request related to a problem? Please describe.**
Make DNA tracing optional as a parameter in the config file.

**Describe the solution you'd like**
When `True`, DNA tracing should run and output the DNA tracing stats into the folder-wise and all_statistics.csv along with the grainstats.

When `False`, it should not run but still output the grainstats files but only for the grain stats.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

ns-rse, 2022-08-02:
Nice, this sort of flexibility could run through all aspects of `process_scan()` so that users can choose whether they run...

* Filters
* Grains
* GrainStats
* DNATracing

Happy to have the other stages as separate issues though.

ns-rse, 2022-08-04:
I've been thinking about this and think we can improve the flow of options through `process_scan()`.

Configuration options come from a YAML configuration file.

```yaml
base_dir: ./tests/
output_dir: ./output
save_plots: true
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
channel: Height
amplify_level: 1.0
filter:
  threshold:
    method: std_dev # Options : otsu, std_dev, absolute
    otsu_multiplier: 1.0
    std_dev: 1.0
    absolute:
      - 1
      - -1
grains:
  gaussian_size: 2.0 
  gaussian_mode: nearest
  absolute_smallest_grain_size: 100
  background: 0.0
  mask_direction: upper # Options: upper, lower (currently only applied to thresh+mask image)
  save_cropped_grains: True
  threshold:
    method: std_dev # Options : otsu, std_dev, absolute
    otsu_multiplier: 1.0
    std_dev: 1.0
    absolute:
      - 1
      - -1
plotting:
  save: True
  zrange: [0, 3] # low and high height range for core images (can take [null, null])
  image_set: core # Options : all, core
  colorbar: True
  cmap: nanoscope # Options : nanoscope, afmhot
```

We can introduce a `run: true # Options : true, false` to each key section (i.e. `filters`, `grains`, `dnatracing` and `plotting`) such that it looks like...

```yaml
base_dir: ./tests/
output_dir: ./output
save_plots: true
warnings: ignore
cores: 4
quiet: false
file_ext: .spm
channel: Height
amplify_level: 1.0
filter:
  run: true # Options : true, false
  threshold:
    method: std_dev # Options : otsu, std_dev, absolute
    otsu_multiplier: 1.0
    std_dev: 1.0
    absolute:
      - 1
      - -1
grains:
  run: true # Options : true, false
  gaussian_size: 2.0 
  gaussian_mode: nearest
  absolute_smallest_grain_size: 100
  background: 0.0
  mask_direction: upper # Options: upper, lower (currently only applied to thresh+mask image)
  save_cropped_grains: True
  threshold:
    method: std_dev # Options : otsu, std_dev, absolute
    otsu_multiplier: 1.0
    std_dev: 1.0
    absolute:
      - 1
      - -1
dnatracing:
  run: true # Options : true, false
plotting:
  run: true # Options : true, false
  save: True
  zrange: [0, 3] # low and high height range for core images (can take [null, null])
  image_set: core # Options : all, core
  colorbar: True
  cmap: nanoscope # Options : nanoscope, afmhot
```

We then modify `process_scan()` to take a dictionary of each set of options and use these to control running and the options used in each stage

```python
def process_scan(
    image_path: Union[str, Path],
    base_dir: Union[str, Path],
    filter_config: dict,
    grains_config: dict,
    dnatracing_config: dict,
    plotting_config: dict,
    channel: str = "Height",
    amplify_level: float = 1.0,
    output_dir: Union[str, Path] = "output",
) -> None:
    ...
# Filter Image :
    #
    # The Filters class has a convenience method that runs the instantiated class in full.
    if filter_config["run"]:
        filtered_image = Filters(
            image_path,
            threshold_method=filter_config["threshold"]["method"],
            otsu_threshold_multiplier=filter_config["threshold"]["otsu_multiplier"],
            threshold_std_dev=filter_config["threshold"]["std_dev"],
            threshold_absolute_lower=filter_config["threshold"]["absolute"][0],
            threshold_absolute_upper=filter_config["threshold"]["absolute"][1],
            channel=channel,
            amplify_level=amplify_level,
            output_dir=filter_out_path,
        )
        filtered_image.filter_image()
    # Find Grains :
    # The Grains class also has a convenience method that runs the instantiated class in full.
    if grains_config["run"]:
        try:
            LOGGER.info(f"[{filtered_image.filename}] : *** Grain Finding ***")
            ...
```

This then simplifies the creation of the `partial()` function used in Multi-processing.

```python
    processing_function = partial(
        process_scan,
        base_dir=config["base_dir"],
        channel=config["channel"],
        amplify_level=config["amplify_level"],
        filter_config=config["filter"],
        grains_config=config["grains"],
        dnatracing_config=config["dnatracing"],
        plotting_config=config["plotting"],
        output_dir=config["output_dir"]
    )
```

What would be really sensible though, would be to adjust the names of fields in the YAML file to match the options for the classes `Filter()`, `Grains()` etc. and we could then flatten each of the dictionaries using something like [this](https://stackoverflow.com/a/6027615) so that we could use `**filter_config` when instantiating `Filters()`.

MaxGamill-Sheffield, 2022-08-04:
@ns-rse, I think this is very sensible and I wonder if it may solve the discrepancies myself and @SylviaWhittle have with the `PLOT_DICT` in #244.

Additionally, I see that some parameters are not used or are duplicates, i.e. line 3 of `example.yaml` is `save_plots` but then there is `plotting['save']` at line 34.

I also think @SylviaWhittle mentioned upper and lower thresholds for `std_dev`, so maybe "config yaml and input revamp" should be a separate issue? What does everyone think?

ns-rse, 2022-08-05:
One of the things I've also thought might be more sensible is to parameterise `PLOT_DICT` and have it as a `plotting_dictionary.yaml` file that is loaded (can be a default included resource in the package).

Certainly need to sanitise/rationalise the configuration file in light of changes and new options being added and as a new issue. The schema above should be a starting point.

PR #258 (2022-08-08): DNA tracing optional
Resolves #241 

Makes DNA tracing optional based on the configuration file option...

```
dnatracing:
  run: true # Whether to run DNA Tracing. Options : true, false
```

If _not_ run then just the grainstats data frame is saved.
closed 2022-08-08 by MaxGamill-Sheffield