AFM-SPM/TopoStats #260: Some user's global PLOT_DICT not updated in main()

MaxGamill-Sheffield, 2022-08-08:
**Describe the bug**
User-dependant outputs were first discovered via PR #244 , where the produced "_processed" plots were not scaled by `zrange` as expected. However, this affects all plotting attributes as default values are selected.

**To Reproduce**
Unable for some to reproduce. For myself (and Sylvia):
1. Clone the following branch (issue apparent on all but this one has print states present at time of writing, showing the issue) `git clone https://github.com/AFM-SPM/TopoStats.git -b maxgamill-sheffield/238-enforce-zrange-and-legend-outside-img`
2. Run: `run_topostats -c config/example.yaml`.
3. See that the produced "_processed" images after running do not have the zrange between 0 and 3nm.
4. Additionally see that the plotting dictionary is not updated via the print statements in this branch.

**Expected behavior**
The `PLOT_DICT` variable called in the `process_scan` function should be the `main` function updated global variable `PLOT_DICT` and not the original. This should then go on to produce images with the correct scale bar as one is now specified in  `plot_and_save(... , **PLOT_DICT[plot_name])`

**Output**
Before updating in `main()`
[image]
After updating in `main()`
[image]
When dictionary is called in `process_scan()`
[image]

As one can see, this resulting behaviour is caused by the global variable `PLOT_DICT` not being correctly updated, and thus, the original dictionary is being invoked when called in the `process_scan` function.

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: MacOS 12.4
 - topostats version: 9ef159c
 - Python version: tested with 3.8 and 3.10.5
 - Environment packages:
cycler==0.11.0
fonttools==4.34.4
imageio==2.21.0
joblib==1.1.0
kiwisolver==1.4.4
matplotlib==3.5.2
networkx==2.8.5
numpy==1.23.1
packaging==21.3
pandas==1.4.3
Pillow==9.2.0
pyparsing==3.0.9
pySPM==0.2.23
python-dateutil==2.8.2
pytz==2022.1
PyWavelets==1.3.0
ruamel.yaml==0.17.21
ruamel.yaml.clib==0.2.6
scikit-image==0.19.2
scikit-learn==1.1.1
scipy==1.9.0
seaborn==0.11.2
six==1.16.0
threadpoolctl==3.1.0
tifffile==2022.7.31
topostats @ file:///Users/Maxgamill/Desktop/Uni/PhD/top/TopoStats
tqdm==4.64.0

**Additional context**
Although my machine is a Mac, Sylvia's machine this was tested on is a windows so this is cross OS.

MaxGamill-Sheffield, 2022-08-08:
Commenting out `run_topostats.py` lines 553-561 (the `Pool` stuff) and replacing with `results = processing_function()`, seems to solve the issue and correctly update the global `PLOT_DICT` within `process_scan`.
Lines adjusted:
[image]
Resulting print statement
[image]

Now, although the full script doesn't run, we can see that removing the multiprocessing part, the updated global dictionary is passed into the partial function `processing_function`.

The question is why is multiprocessing breaking this?

MaxGamill-Sheffield, 2022-08-08:
This short tests shows the issue on my side:
```
from multiprocessing import Pool

d = {'a':0}

def func(x):
    'Defines psudo func to just check the global dict'
    print("process: ", d)
    return x*x

def main():
    print('main: ', d)
    d['b']=1
    print('main: ', d)
    with Pool(processes=4) as pool:
        for i in pool.imap_unordered(func, range(10)):
                pass

if __name__ == "__main__":
    main()
```

MaxGamill-Sheffield, 2022-08-08:
Found this on SO:

Multiprocessing Pools fork (or spawn in a way intended to mimic forking on Windows) its worker processes at the moment the Pool is created. forking maps the parent memory as copy-on-write in the children, but it doesn't create a persistent tie between them; after the fork, changes made in the parent are not visible in the children, and vice versa. **You can't use any variables defined after the Pool was created, and changes made to variables from before the Pool was created will not be reflected in the workers.**

**Typically, with a Pool, you want to avoid mutable global state entirely; have all the data needed passed to the function you're imap-ing** (or whatever) as arguments (which are serialized and sent to the children, so the state is correct), and have the function return any new data instead of mutating globals, which serializes it and sends it back to the parent process to use as it sees fit.

Managers are an option, but not usually the correct option with Pools; you usually want to stick to workers only looking at read only globals from before the Pool was created, or working with arguments and returning new values, not using global state at all.

MaxGamill-Sheffield, 2022-08-08:
Moving `PLOT_DICT` to be passed into the `process_scan` func from `main()` resolves this. 

Will add into an open push request.

ns-rse, 2022-08-15:
Good digging and apologies for causing the head-scratching in the first place structuring it this way.

It may be worthwhile combining and addressing with #255 which suggests that the plotting dictionary could be paramterised (i.e. made slightly more configurable rather than being hard-coded in `run_topostats.py`). This would mean setting all of the plotting options (currently defined in `PLOT_DICT`) in the `config.yaml`, nested under the `plotting` section.

MaxGamill-Sheffield, 2022-08-15:
> Good digging and apologies for causing the head-scratching in the first place structuring it this way.
>
> It may be worthwhile combining and addressing with #255 which suggests that the plotting [...]

@ns-rse Do you think this should be included in PR #244 then? or as a separate PR to avoid doing to many things in one - similar to how I found myself in this mess in the first place

ns-rse, 2022-08-15:
Hi @MaxGamill-Sheffield ,

I'd keep #244 as it is now and then we can focus on restructuring the config and paramterising `PLOT_DICT` under the `config.yaml` in a more focused branch.

I'll get #244 reviewed shortly.

closed 2022-08-15 by MaxGamill-Sheffield