AFM-SPM/TopoStats #269: Backgrounding should be moved from Grains class to Filters class

ns-rse, 2022-08-23:
Currently backgrounding is performed in the Grains class but should really be done in the Filter class.

ns-rse, 2022-08-23:
This can be tied into #232

MaxGamill-Sheffield, 2022-09-02:
This does not do what we initially thought it does. `background` is used as an input to [sklearn.morphology.label](https://scikit-image.org/docs/dev/api/skimage.morphology.html#skimage.morphology.label) which "Consider all pixels with this value as background pixels, and label them as 0. By default, 0-valued pixels are considered as background pixels."

Therefore, my mind goes to how images such as in silicon wafers or lipid bilayers would handle this as there is no "background" unless we have already labeled all non-grain pixels as 0, then this would be the default anyway. 
More discussion needed I think.

MaxGamill-Sheffield, 2022-09-13:
After further discussion, it was generally accepted that as we split upper and lower grains, the background (by default) will always be labeled as `0` and grains as `1, 2, 3...`. This means that the `background` value in the config can be removed as this should always be 0.

closed 2022-09-13 by MaxGamill-Sheffield