AFM-SPM/TopoStats #277: Move cropped grain image creation from grainstats to plotting

MaxGamill-Sheffield, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
Currently, the cropped grain production is located inside `Grainstats` to to its proximity to calculating the grain centres. This however, is a little odd and might seem better suited to somewhere like plotting if coordinates are supplied.

**Describe the solution you'd like**
Either:
1. Read from the results file the grain centres, convert to pixels and then undergo cropping and plotting, possibly with cropping in utils.
2. Output the grain centres from grainstats to feed into a `utils.py` function to identify the region to crop and then into `plottingfuncs.py` to  plot

**Describe alternatives you've considered**
No others

**Additional context**
Add any other context or screenshots about the feature request here.

ns-rse, 2022-10-05:
I think once I get #295 sorted this will follow naturally as my intention is to make the grains isolated in grainstats an attribute of the `GrainStats` class which will then make it available for plotting elsewhere.

ns-rse, 2023-01-06:
I think this is a duplicate of #138

closed 2023-01-06 by ns-rse