AFM-SPM/TopoStats #294: dntracing to use bounded grains from GrainStats

ns-rse, 2022-09-23:
Currently dnatracing takes the labelled image and extracts each identified molecule but then works with the whole 1000x1000 pixel image. It would be (marginally) faster and make more sense if instead dnatracing worked with cropped images.

We already have existing code that does this in `topostats/grainstats.py` as lines 162-276 loops over all grains (`region_properties`) and crops them before plotting and calculating statistics.

To this end rather than duplicate code we can modify `grainstats/GrainStats` class as described in

MaxGamill-Sheffield, 2022-12-06:
Update on this issue, while refactoring dnatracing.py, I've seen that the only reason we process grain-by-grain is due to a dilation step. I'm hoping that this step is the source of some poor tracing issues and that if it can be removed, the skeletonisation and pruning steps to follow can be done across the whole image (currently this would save about 0.8s on minicircles.spm).

After, I believe all the current statistics outputs use the list of points produced by the traces for their calculations.

This means that bounded grains may not need to be used at all.

MaxGamill-Sheffield, 2022-12-16:
I also see that many of the subsequent steps in dnatracing require the functions to not "hit the edges". This means that we'd need to think about making the bounding boxes bigger for input here.

PR #600 (2023-06-12): Improves processing of multiple grains during dnatracing
Closes #289
Closes #294 

This removes processing of a whole image and instead makes the unit of processing in dnatracing a single grain. Processing multiple grains is done via a helper function `trace_image()` which calls `trace_grain()` the later is a thin wrapper to the `traceDna()` class.

When cropping the image and masks for processing they are padded twice compared to the scikit-image determined bounding box.

1. The original image/mask are paded by a user configurable amount. This is so that heights which will be used when height [... cut]
closed 2023-06-12 by ns-rse