AFM-SPM/TopoStats #298: Improve flattening

SylviaWhittle, 2022-09-27:
**Describe the bug**
Flattening is not quite as good as we require. We propose to create a set of tests to determine the quality of, and diagnose problems in flattening.

- Further tests for flattening
- Demos of what it can and cannot currently flatten
- Add the 'flatten base' feature from Gwyddion. (http://gwyddion.net/documentation/user-guide-en/leveling-and-background.html) this paper has lots of information on it too: (https://www.beilstein-journals.org/bjnano/content/pdf/2190-4286-3-84.pdf)

Feel free to assign yourself too - I don't want to steal all the fun tasks!

MaxGamill-Sheffield, 2022-10-03:
I've made a little comparison table of the 3 flattening procedures between Old TS, what's in the TopoStats paper, and what we have at the minute in new TopoStats if this helps with the procedure:
[image]

MaxGamill-Sheffield, 2022-10-04:
I had a meeting with Alice where she has suggested the following instead of flatten base as this seemed to work well on images within Gwyddion:
- Align rows via Median of Differences
- Level
- Mask
- Align rows viaMedian of Differences
- Level
- Re-mask
- Align rows via Median

ns-rse, 2022-11-15:
Closed with #351

closed 2022-11-15 by ns-rse

MaxGamill-Sheffield, 2023-05-22:
Removed gdrive test data