AFM-SPM/TopoStats #300: Support JPK formats

ns-rse, 2022-10-04:
As we work towards supporting more fileformats (see #137) we have decided to prioritise being able to load and process JPK file formats. The [afmformats](https://github.com/AFM-analysis/afmformats/) package supports this format and so the `topostats.io` sub-module should be expanded to support this.

Currently loading of Bruker formats is done via `topostats.io.load_scan()` to avoid breaking functionality this should be converted to (another!) Factory Method that calls the appropriate routine for loading the given file format.

**Describe the solution you'd like**

Be able to load JPK files.

**Describe alternatives you've considered**
N/A

MaxGamill-Sheffield, 2022-10-12:
I've tried renaming the extension (as I recall it came with none) to all the JPK formats above but it still couldn't open. 

I get the message "jpk format not supported" when the file extension is `.jpk`, and the message "Bad zip file" when copying the extension above. This refers to the JPK file structure as it is actually a zipped archive with a top level header folder and subfolders containing data.

I have tried with another JPK loading library (https://gitlab.gwdg.de/ikuhlem/jpkfile) with the same things but I also get "Bad zip file".

However, Gwyddion (grrr) is able to load this with no issues, and can tell it is a jpk file:
[image]

So I'm at a bit of a loss

ns-rse, 2022-10-12:
I have two files the original I got from the shared drive...

`force-save-2019.01.11-12.23.32.518.jpk-force` 

...this is a zip archive of sorts (Google Drive recognises it as such).

```
import jpkfile

sample_jpk = jpkfile.JPKFile("force-save-2019.01.11-12.23.32.518.jpk-force")
/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/jpkfile/jpkfile.py:92: UserWarning: Did not find encoder parameters for channel smoothedCapacitiveSensorHeight!
  warnings.warn("Did not find encoder parameters for channel {}!".format(split[3][:-4]))
/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/jpkfile/jpkfile.py:92: UserWarning: Did not find encoder parameters for channel smoothedMeasuredHeight!
  warnings.warn("Did not find encoder parameters for channel {}!".format(split[3][:-4]))
 dir(sample_jpk)
Out[9]: 
['__class__',
 '__delattr__',
 '__dict__',
 '__dir__',
 '__doc__',
 '__eq__',
 '__format__',
 '__ge__',
 '__getattribute__',
 '__gt__',
 '__hash__',
 '__init__',
 '__init_subclass__',
 '__le__',
 '__lt__',
 '__module__',
 '__ne__',
 '__new__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 '__weakref__',
 'archive_type',
 'data',
 'get_array',
 'get_info',
 'has_shared_header',
 'jpk_zip',
 'num_segments',
 'parameters',
 'read_files',
 'segments',
 'shared_parameters']
sample_jpk.data

In [11]: sample_jpk.archive_type
Out[11]: ''

In [12]: sample_jpk.parameters
Out[12]: 
{'date': datetime.datetime(2019, 1, 11, 13, 49, 53),
 'jpk-data-file': 'spm-forcefile',
 'file-format-version': '2.0',
 'file-format-features': '4000013',
 'type': 'force-scan-series',
 'force-scan-series': {'force-segments': {'count': '2'},
  'header': {'type': 'xy-position-force-scan-series-header',
   'position': {'x': '1.183529859504612E-13', 'y': '-3.637674649418815E-13'},
   'position-index': '0',
   'force-settings': {'type': 'relative-force-settings',
[... 478 lines cut ...]
     'type': 'simple',
     'comment': '',
     'base-calibration-slot': 'volts',
     'calibration-slot': 'nominal',
     'scaling': {'type': 'linear',
      'style': 'offsetmultiplier',
      'offset': '1.4994616934432393E-5',
      'multiplier': '-1.3103859706933053E-7',
      'unit': {'type': 'metric-unit', 'unit': 'm'}}},
    'calibrated': {'name': 'Calibrated height',
     'defined': 'true',
     'type': 'file',
     'file': '/home/nw3bio/jpkdata/nic/2018_11_26 long z cal/H-06-0126.cal',
     'comment': 'Nic_26/11/18\\nXinyue 24/08/18\\nNic&Xinyue 30/04/18\\nNic 27/06/2017',
     'base-calibration-slot': 'nominal',
     'calibration-slot': 'calibrated',
     'scaling': {'type': 'linear',
      'style': 'offsetmultiplier',
      'offset': '0.0',
      'multiplier': '0.74199',
      'unit': {'type': 'metric-unit', 'unit': 'm'}}}}}})
```
Can access more based on keys/names and so forth.

Don't know if any of this is any use :shrug:

MaxGamill-Sheffield, 2022-10-13:
The issue here is as you found, its a 1D force measurement and not an image which would be used with TopoStats.

I've done a little more investigating on the Gwyddion side of things and looked at their [import list](http://gwyddion.net/documentation/user-guide-en/file-formats.html) which shows that a `.jpk` file *is* a thing and that's what I believe is being loaded (or `.jpk-qi-image` seems like a likely candidate too). 

So, it seems like the libraries we've looked into don't support just all JPK file extensions which is an issue

[image]

MaxGamill-Sheffield, 2022-10-13:
@ns-rse, @alicepyne has suggested to look into these possible libraries to load JPK files:
[Hyperspy](https://github.com/hyperspy/hyperspy)
Although these are the supported Hyperspy formats:
Format | Read | Write | lazy
-- | -- | -- | --
Gatan’s dm3 | Yes | No | Yes
Gatan’s dm4 | Yes | No | Yes
FEI’s emi and ser | Yes | No | Yes
hspy | Yes | Yes | Yes
zspy | Yes | Yes | Yes
Image: e.g. jpg, png, tif, … | Yes | Yes | Yes
TIFF | Yes | Yes | Yes
MRC | Yes | No | Yes
MRCZ | Yes | Yes | Yes
EMSA/MSA | Yes | Yes | No
NetCDF | Yes | No | No
Ripple | Yes | Yes | Yes
SEMPER unf | Yes | Yes | Yes
Blockfile | Yes | Yes | Yes
DENSsolutions’ Impulse log | Yes | No | No
DENSsolutions’ Digiheater log | Yes | No | No
Bruker’s bcf | Yes | No | Yes
Bruker’s spx | Yes | No | No
EMD (NCEM) | Yes | Yes | Yes
EMD (Velox) | Yes | No | Yes
Protochips log | Yes | No | No
EDAX spc and spd | Yes | No | Yes
h5USID h5 | Yes | Yes | Yes
Phenom elid | Yes | No | No
DigitalSurf’s sur and pro | Yes | No | No
Nexus nxs | Yes | Yes | Yes
EMPAD xml | Yes | No | Yes
JEOL asw, map, img, pts, eds | Yes | No | No
TVIPS .tvips | Yes | Yes | Yes

[image-analysis](https://github.com/AFM-SPM/image-analysis)
Which is written in matlab and I cannot find where they import the files.

[PyDrive](https://github.com/googlearchive/PyDrive)
Which seems like a client to connect python with Google drive and has no way of opening AFM images

Otherwise, Alice has said that she can put us into contact with the head of JPK who might be able to help us which may seem like the way to go. 

If we are then able to open additional JPK filetypes, it might be worth having a in-house solution that we can contribute to `afmformats` and then remove from our codebase once the addition is merged with their code.

PR #347 (2022-11-23): Maxgamill sheffield/#300 jpk
Closes #300

This PR builds off the `.asd` branch due to the change in input affecting how these files are processed, so it might be best for that to be merged first.

As images are in a `TIFF` format, they are loaded using `tifffile` which comes installed with `scikit-image`.

The jpk file specification document is used to translate the hexadecimal tiff tags and obtain the values needed to process this filetype. These are the: channel, trace/retrace, real and px lengths, and the image.

Tests have also been made for this filetype excluding the px_to_nm scaling fun [... cut]
closed 2022-11-23 by MaxGamill-Sheffield