AFM-SPM/TopoStats #302: Improve validation of configuration

ns-rse, 2022-10-05:
#299 addresses the need for the validation of our YAML input configuration and has been addressed by `INSERT`.

However, not all values are complete as valid ranges for each of the fields are not something @ns-rse can provide. Now that the framework is in place it should be refined and the purpose of this ticket is to record discussion of valid ranges for each of the fields prior to implementing.

MaxGamill-Sheffield, 2022-10-11:
> #299 addresses the need for the validation of our YAML input configuration and has been addressed by `INSERT`.
> 
> However, not all values are complete as valid ranges for each of the fields are not something @ns-rse can provide. Now that the framework is in place it should be refined and the purpose of this ticket is to record discussion of valid ranges for each of the fields prior to implementing.
> 
> Fields that are yet to have ranges defined...
> 
> Field	Type	Current value in `config/example.yaml`
> `filter.amplify_level`	`float`	`1.0`
> `filter.otsu_threshold_multiplier`	`float`	`1.0`
> `filter.threshold_absolute_lower`	`float`	`-1.0`
> `filter.threshold_absolute_upper`	`float`	`1.0`
> `filter.gaussian_size`	`float`	`0.5`
> `grains.absolute_smallest_grain_size`	`int`	`100`
> `grains.otsu_threshold_multiplier`	`float`	`1.0`
> `grains.otsu_threshold_multiplier`	`float`	`1.0`
> `grains.threshold_absolute_lower`	`float`	`1.0` **NB** Differs from `filter.threshold_absolute_lower`, was changed in [284](https://github.com/AFM-SPM/TopoStats/pull/284/files#diff-64a1933b775fbf7d3dc5f338c5b2fdb6d334aa1401172abc10f8bcb9184f6715) but not in `topostats/default_config.yaml`.
> `grains.threshold_absolute_upper`	`float`	`1.0`
> `grains.absolute_area_threshold.upper`	`int`	`500-800`
> `grains.absolute_area_threshold.lower`	`int`	`null-null`
> `grains.background`	`float`	`0.0`
> `grainstats.cropped_size`	`float`	`40.0`
> `plotting.save_format`	`str`	`png` but not sure we need to test these, Matplotlib will handle
> `plotting.zrange`	`int`	`0-3`

Hi @ns-rse, looking through this list, most of these fields don't need ranges or values as this is upto the user (only some general catching like `grains.absolute_area_threshold_upper` should be a list of positive floats, although now this should be a dict of upper and lower of lists of positive floats after the area threshold merge). 

I also think that the config options `filter.amplify_level` and `grains.background` should not be included as I'm unsure of their function - amplify level seems to modify the heights out of the AFM resulting in unrepresentitive data, and the background seems to always become 0 as it looks like it defines what number the background is for the mask in a scipy/scikit_image function.

But yes I think other than that and the fields you have pointed out, these do not need any specified values / value ranges.

I hope this helps

ns-rse, 2022-10-21:
> I also think that the config options filter.amplify_level and grains.background should not be included as I'm unsure of their function - amplify level seems to modify the heights out of the AFM [...]

Both of these configuration options have been removed see #288 and #310.

ns-rse, 2022-11-30:
We now have validation in place, although on reflection I think it could now do with some refactoring to have a single function that performs validation and then define individual Schemas which are passed in.

ns-rse, 2023-10-20:
#384 refactored validation to a function which took an schema as an argument so closing this.

closed 2023-10-20 by ns-rse