AFM-SPM/TopoStats #312: Adai - Matching Trace Images

MaxGamill-Sheffield, 2022-10-11:
**Is your feature request related to a problem? Please describe.**
AFM's produce 2 "images" for each channel, one is from a forward trace and the other the backward trace as the AFM tip raster scans across the image. 

Due to a number of reasons, these traces may not perfectly align, and thus the images can look "out of sync".

**Describe the solution you'd like**
A feature which can align the two images produced.

**Additional context**
My research (Max):

This issue is a complex one but usually goes under the name of image alignment/registration. The most applicable and I think easiest to implement is via Homography for which there is an `openCV` module for. This method works to not only account for offsets but also adjusts warped images and has a number of steps:
1. Identify features in both images
2. Convert features to points 
3. Identify very similar points in both images via correlation and discard the rest (must have > 3 points)
4. Use the points to calculate the Homography matrix (https://en.wikipedia.org/wiki/Homography)
5. Apply the matrix to the image

ns-rse, 2023-11-10:
Is this still an issue @MaxGamill-Sheffield ?

Is aligning the pictures required and if so how does it fit into the workflow?

MaxGamill-Sheffield, 2023-11-10:
I guess maybe more of a feature request for the future but aligning both images could possibly help improve resolution