AFM-SPM/TopoStats #319: Reorder the 'grains' section of the config file

Jean-Du, 2022-10-12:
**Is your feature request related to a problem? Please describe.**
At the moment, the 'grains' section contains options for both height-based selection and area-based selection, and it is not very clear which one each line refers to, and this could cause confusion to users unfamiliar with the project.

**Describe the solution you'd like**
Reorder the parameters in the 'grains' section of the config so that all height-based entries are in one place and all area-based entries are in another, with clearer commenting showing which one is which.

**Describe alternatives you've considered**
I considered creating nested subsections within the 'grains' section, but this would cause unecessary complications. 

**Additional context**
Add any other context or screenshots about the feature request here.

MaxGamill-Sheffield, 2022-10-12:
Just incase you've missed some of the recent updates, there is no example.yaml config file anymore. 

These changes should instead be made to `topostats/default_config.yaml`.

Jean-Du, 2022-10-12:
@MaxGamill-Sheffield Thanks a lot! I'll keep that in mind

ns-rse, 2022-10-17:
Close by #328

PR #328 (2022-10-17): Reordered the 'grains' section of the config file
Reordered the 'grains' section of the config file so that it's clearer to users

closes #319
closed 2022-10-17 by ns-rse