AFM-SPM/TopoStats #329: Empty all_statistics.csv when some grains are too small in area

Jean-Du, 2022-10-17:
**Describe the bug**
When the 'absolute_smallest_grain_size' is set to 0, the 'all_statistics.csv' file becomes empty.

**To Reproduce**

- Command used to run topostats: `run_topostats --config config.yaml`
- Config file:
https://drive.google.com/file/d/1c4i_9sC3zbF2b7A8uqI6jxCUBpf4t2Vx/view?usp=sharing
-The spm file used:
https://drive.google.com/file/d/1lpBqI_azZHDBmMkD5WVlo4lMh8kceVwU/view?usp=sharing

**Expected behavior**
A non-empty all_statistics.csv. If some grains are too small for statistics to be calculated, they should just be skipped and the other grains that are big enough should still be on the csv file.

**Output**

INFO:topostats:[filename] : Errors occurred attempting to calculate grain statistics and DNA tracing statistics
[INFO    ] [topostats] Unable to generate folderwise statistics as 'all_statistics.csv' is empty

**(Not required if using Docker) Your computer configuration (please complete the following information):**
 - OS: Windows 10 Pro version 21H2
 - topostats version: d4e10a9
 - Python version: 3.10.
 - Optionally, your installed packages: best way to get this is to paste the results of typing `pip freeze`.

**Additional context**
Add any other context about the problem here.

ns-rse, 2023-02-13:
I've just checked processing the file linked from here using  the current `HEAD` of `main` branch (commit `7f866c1`) and `all_statistics.csv` shown below is produced.

```
image,threshold,molecule_number,centre_x,centre_y,radius_min,radius_max,radius_mean,radius_median,height_min,height_max,height_median,height_mean,volume,area,area_cartesian_bbox,smallest_bounding_width,smallest_bounding_length,smallest_bounding_area,aspect_ratio,max_feret,min_feret,contour_lengths,circular,end_to_end_distance,basename
20220411_3ng_TEL40_3mMNicl2_0_00000.0_00022,upper,0,3.0619734480574324e-07,3.25658519847973e-08,2.6469182857527876e-09,2.279547035564762e-08,1.2798577768992227e-08,1.2817266756267162e-08,6.342968486322113e-10,4.133697662960117e-09,2.3308484105561402e-09,2.2531640213810523e-09,7.791493169334122e-25,3.4580230712890637e-16,9.813308715820315e-16,1.0515859038450643e-08,4.405443879841588e-08,4.632702684221924e-16,4.189333333333333,4.407457118699379e-08,1.0515859038450641e-08,4.434027293244973e-08,False,3.547449813175932e-08,/home/neil/work/projects/topostats/tmp/small_grains/data
20220411_3ng_TEL40_3mMNicl2_0_00000.0_00022,upper,1,5.411696549728609e-08,1.1625284363026822e-07,1.64847824041453e-09,5.815917396042979e-08,3.3009394814738884e-08,3.3640018252281545e-08,6.369997391160268e-10,3.227280961601312e-09,1.5888260879897012e-09,1.5716975940929218e-09,1.1501564329636219e-24,7.31792449951172e-16,6.0562133789062515e-15,2.0402644230769233e-08,1.1542217548076924e-07,2.3549175826755508e-15,5.65721649484536,1.1543224867611231e-07,2.0402644230769233e-08,1.247672511976109e-07,False,1.0182793776416057e-07,/home/neil/work/projects/topostats/tmp/small_grains/data
20220411_3ng_TEL40_3mMNicl2_0_00000.0_00022,upper,2,1.7164587566111972e-07,1.5989603489422084e-07,6.2175741537419845e-09,6.533276225553173e-08,3.146413324032906e-08,3.111900820988222e-08,6.351273049991444e-10,4.550830277689065e-09,1.8183095024257983e-09,1.8984248404842586e-09,1.719267079373037e-24,9.056282043457034e-16,6.915111541748049e-15,3.537081199832281e-08,1.1403782421402461e-07,4.033610440972049e-15,3.2240657698056796,1.1404548413585797e-07,3.537081199832281e-08,1.9766556450072866e-07,False,1.0318869733202411e-07,/home/neil/work/projects/topostats/tmp/small_grains/data
20220411_3ng_TEL40_3mMNicl2_0_00000.0_00022,upper,3,2.4875082951254826e-07,2.931599149372587e-07,3.4967465029506687e-09,6.532629934917624e-08,3.153605346923056e-08,3.1674834213590406e-08,6.35600285807563e-10,8.880550716025494e-09,1.6499740131364661e-09,1.9764248066231346e-09,1.6744580320928522e-24,8.472156524658205e-16,5.6496620178222674e-15,3.8934179271217896e-08,1.0695972963026113e-07,4.1643892882255836e-15,2.7471936389148737,1.0696003347506175e-07,3.893417927121789e-08,1.5706879679039918e-07,False,1.0163501205017136e-07,/home/neil/work/projects/topostats/tmp/small_grains/data
```

From this masked image

[image]

@Jean-Du would you be able to check this is what is expected and we can then squash another :lady_beetle: and move a step closer to a new release :crossed_fingers:

ns-rse, 2023-02-13:
Ah, thanks @Jean-Du I forgot to set the `smallest_grain_size` as I knew the config file has changed and it wouldn't validate and so just ran it with the defaults.

I'll look again. Thanks for the quick feedback and sorry for the mistake.

ns-rse, 2023-02-15:
Since implementing validation the check made is (see [line 137 of `topostats/validation.py`](https://github.com/AFM-SPM/TopoStats/blob/main/topostats/validation.py#L137))...

```python
            "smallest_grain_size_nm2": lambda n: n > 0.0,
```

Thus if a configuration file has...

```yaml
  smallest_grain_size_nm2: 0
```

This is caught by the validation step and throws the error....

```
❱ run_topostats --config test.yaml
Traceback (most recent call last):
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/schema.py", line 405, in validate
    nvalue = Schema(svalue, error=e, ignore_extra_keys=i).validate(value, **kwargs)
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/schema.py", line 463, in validate
    raise SchemaError(message, e.format(data) if e else None)
schema.SchemaError: <lambda>(0) should evaluate to True

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/schema.py", line 405, in validate
    nvalue = Schema(svalue, error=e, ignore_extra_keys=i).validate(value, **kwargs)
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/schema.py", line 409, in validate
    raise SchemaError([message] + x.autos, [e.format(data) if e else None] + x.errors)
schema.SchemaError: Key 'smallest_grain_size_nm2' error:
<lambda>(0) should evaluate to True

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/neil/work/projects/topostats/TopoStats/topostats/validation.py", line 27, in validate_config
    schema.validate(config)
  File "/home/neil/.virtualenvs/topostats/lib/python3.10/site-packages/schema.py", line 409, in validate
    raise SchemaError([message] + x.autos, [e.format(data) if e else None] + x.errors)
schema.SchemaError: Key 'grains' error:
Key 'smallest_grain_size_nm2' error:
<lambda>(0) should evaluate to True

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/neil/.virtualenvs/topostats/bin/run_topostats", line 8, in <module>
    sys.exit(main())
  File "/home/neil/work/projects/topostats/TopoStats/topostats/run_topostats.py", line 132, in main
    validate_config(config, schema=DEFAULT_CONFIG_SCHEMA, config_type="YAML configuration file")
  File "/home/neil/work/projects/topostats/TopoStats/topostats/validation.py", line 30, in validate_config
    raise SchemaError(
schema.SchemaError: There is an error in your YAML configuration file configuration. Please refer to the first error message above for details
```

This raises the question...

* Will grains of size 0.0nm^2 ever need to be detected?

If so the validation needs correcting as well as handling the original error.

If not then the original error could be considered dealt with since the validation process ensures a value of `smallest_grain_size_nm2: 0` is not permitted.

SylviaWhittle, 2023-02-15:
I don't think that setting this to zero would ever be needed.

ns-rse, 2023-02-16:
@Jean-Du what do you think? You tested this with a value of zero, what was your reason for doing so? Will such a value ever be required in the future? If not then I think we can safely close this as the validation highlights this as an erroneous value in the configuration.

Jean-Du, 2023-02-16:
@ns-rse I tested it with 0 initially because I wanted to pick up all the small features. But I could do the same by setting it to 1, so I am happy with this solution. However if this isn't done already, it might be worth adding a comment in the config file explaining that the value cannot be set to 0.

ns-rse, 2023-02-16:
Good point @Jean-Du  #472 does so and also mentions it in the documentation.

closed 2023-02-17 by ns-rse