AFM-SPM/TopoStats #340: Add warning outputs for bad input images

SylviaWhittle, 2022-10-24:
In some cases where users are attempting to flatten poor, or erroneous images, it would be beneficial to be able to warn them that some of their output images will be problematic or lower quality. 

@ns-rse suggested that this could be done by adding a `@attr` to the `Filters()` class and add this to the `grainstats_df` dataframe such that at the end, a script, proposed to be called `warnings.py` could recognise that there were issues and warn the user appropriately though the terminal output. 

Copying @ns-rse 's suggested implementation of the error message:

```
if results["grain_span"].any():
LOGGER.info("#################################################################")
    LOGGER.info("##                                                             ##")
    LOGGER.info("##                     !!!!!!WARNING!!!!!!                     ##")
    LOGGER.info("##                                                             ##")
    LOGGER.info("## Some images have grains spanning the whole image.           ##")
    LOGGER.info("## This causes problems with flattening.                       ##")
    LOGGER.info("##                                                             ##")
    LOGGER.info("## The affected images are listed below, please review         ##")
    LOGGER.info("## carefully                                                   ##")
    LOGGER.info("##                                                             ##")
    LOGGER.info("#################################################################\n\n")
    LOGGER.info(f"Affected files :\n{results[results['grain_span'] == True]['Image Name']}")
```

Having a `warnings.py` file instead of directly implementing this into `run_topostats.py` will reduce bloat and allow for more custom error and diagnostic messages for the users that can have tests written for them.

ns-rse, 2023-10-13:
@SylviaWhittle : Do we still want to do this? In one regard its technically easy to add, the harder part would be how to decide on what constitutes a "bad input image" on which to trigger displaying such a message.

SylviaWhittle, 2023-10-17:
Let's close this until someone asks for it 👍

closed 2023-10-17 by SylviaWhittle