AFM-SPM/TopoStats #349: Rename 'plotting' in the config_file to 'image_output'

Jean-Du, 2022-11-03:
**Is your feature request related to a problem? Please describe.**
The name 'plotting' in the config file is easily confused with the plotting of statistics. It should be renamed to something like 'image_output'.

**Describe the solution you'd like**
Rename 'plotting' to 'image_output'

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ns-rse, 2022-11-09:
I'm in two minds about this as to me its all just "plotting" using Matplotlib (or Seaborn which is just another layer on top), but I can also see why it may confuse.

I've purposefully therefore used `TopoSum` as the class name for summarising the plotting the molecule statistics and have a separate configuration file (I recall talk of it being intimidating having lots of options but when lots of options are what is required then lots of configuration is required!).

As its all "plotting" it might be prudent to group and used the nested structure of YAML/dictionaries to have options for different types of plotting under sub-headings.

If the change does go ahead then the files that will need changing are...

* `topostats/default_config.yaml`
* `topostats/run_topostats.py`

Then possibly some of the code in the branch I'm working on ns-rse/227-plotting.

* `topostats/summary_config.yaml`.

Its not too big a job  to make these changes, but I'd like to get the larger jobs of refactoring plotting and DNATracing addressed.

Jean-Du, 2022-11-10:
The main issue is that for the experimantalists used to old TopoStats, the word 'plotting' refers only to the plotting of data distribution, as the rest is more about image processing. 
I understand that you have more important tasks to deal with, so I can make this change when I get some time to spare.

alicepyne, 2022-11-10:
I think this is one for a discussion before we change the name, as plotting could be pictures or graphs!

ns-rse, 2023-11-10:
Closing, no discussion has been had and we now separate configuration of options for plotting scans out to a `matplotlibrc` params file (see #707) and retain options for plotting distributions in the configuration which helps avoid confusion over how/where to configure options.

closed 2023-11-10 by ns-rse