AFM-SPM/TopoStats #350: Add Molecule Numbers to Plot Option

MaxGamill-Sheffield, 2022-11-04:
**Is your feature request related to a problem? Please describe.**
For single molecule analysis, it might be useful to add the molecule number to the plots (and maybe have this as a binary option for the config too) so people know which molecule produced the analyses in question.

**Describe the solution you'd like**
The following code can produce a text-box like that in the image below that would ideally be good for the processed_mask image.
```ax[0].text(20, 50, "1", bbox=dict(facecolor='white', alpha=0.7))```
[image]

Obviously, the `text(x, y, string...)` should have `x` and `y` derive from the molecules bounding boxes (with an offset that is check to see whether it remains in the image).

MaxGamill-Sheffield, 2024-11-13:
Would require:
• Passing bbox coords into the plotting func (this could then position in the middle for now or be used to find the best (lowest height area) position)
• Flag to set this on or not

MaxGamill-Sheffield, 2024-11-13:
For each of the plotting stages which create unique output files with labels (i.e. grain / branch / mol number), for the key plotting image, add an option to add the object number to the images so easily identify them.

ns-rse, 2025-12-18:
Closed by #1230