AFM-SPM/TopoStats #352: Make tests fully reproducible

ns-rse, 2022-11-09:
## Describe the bug

Its been observed that some of the tests pass when run in isolation but then fail when run as part of the whole suite, e.g.

### This fails

```bash
pytest tests/test_filter_minicircle.py
❱ pytest tests/test_filters_minicircle.py
=============================================================================== test session starts ================================================================================
platform linux -- Python 3.10.8, pytest-7.1.2, pluggy-1.0.0
Matplotlib: 3.6.0
Freetype: 2.6.1
rootdir: /home/neil/work/projects/topostats/TopoStats, configfile: pyproject.toml
plugins: mpl-0.16.1, typeguard-2.13.3, regtest-1.5.0, cov-3.0.0, forked-1.4.0, xdist-2.5.0
collected 10 items                                                                                                                                                                 

tests/test_filters_minicircle.py FF...FFFFF     
...
        ============================================================================= short test summary info ==============================================================================
FAILED tests/test_filters_minicircle.py::test_align_rows_unmasked
FAILED tests/test_filters_minicircle.py::test_remove_x_y_tilt_unmasked
FAILED tests/test_filters_minicircle.py::test_get_mask
FAILED tests/test_filters_minicircle.py::test_align_rows_masked
FAILED tests/test_filters_minicircle.py::test_remove_x_y_tilt_masked
FAILED tests/test_filters_minicircle.py::test_average_background
FAILED tests/test_filters_minicircle.py::test_gaussian_filter
=========================================================================== 7 failed, 3 passed in 18.10s ===========================================================================

### This passes
```
pytest -n14
```

A lot of the differences appear to be in the images and so updating those with `--mpl-generate-path` solves for the fails and then fails for the other set.

## Expected behavior
All tests should pass regardless of the order in which they are run.

## Thoughts

I think this may be down to the fixtures used in `tests/conftest.py`

Currently there is...

```python 
import numpy as np
RNG = np.random.default_rng(seed=1000)
...
@pytest.fixture
def image_random() -> np.ndarray:
    """Random image as NumPy array."""
    rng = np.random.default_rng(seed=1000)
    return rng.random((1024, 1024))

@pytest.fixture
def small_array() -> np.ndarray:
    """Small (10x10) image array for testing"""
    return RNG.random(SMALL_ARRAY_SIZE)

@pytest.fixture
def small_mask() -> np.ndarray:
    """Small (10x10) mask array for testing."""
    return RNG.uniform(low=0, high=1, size=SMALL_ARRAY_SIZE) > 0.5
```

This means that whilst fixtures are cached, what has been returned is a [`Generator`](https://numpy.org/devdocs/reference/random/generator.html) rather than a fixed array that is cached and reused.

Thus on each use of `image_random` / `small_array` / `small_mask` it _generates_ a new array rather than using the one that was generated and cached as a fixture.

## A simple example to test this

`tests/conftest.py`
```python
import pytest
import numpy as np

RNG = np.random.default_rng(seed=1000)

ARRAY_SIZE = 10

@pytest.fixture
def return_generator(array_size: int = ARRAY_SIZE) -> np.ndarray:
    """Return a random generator array of given size.

    Parameters
    ----------
    array_size: int
        Size of array.

    Returns
    -------
    np.ndarray
        Numpy array of array_size.
    """
    return RNG.random(array_size)

@pytest.fixture
def return_array(array_size: int = ARRAY_SIZE) -> np.ndarray:
    """Return an array randomly generated numbers of given size.

    Parameters
    ----------
    array_size: int
        Size of array.

    Returns
    -------
    np.ndarray
        Numpy array of array_size.
    """
    random_array = RNG.random(array_size)
    return random_array
```

`tests/test_a_theory.py`
```python
def test1(return_generator, return_array):
    print(f"generator   : {return_generator}")
    print(f"array       : {return_array}")
    assert False

def test2(return_generator, return_array):
    print(f"generator   : {return_generator}")
    print(f"array       : {return_array}")
    assert False
```

Run the tests...

```bash
 ❱ pytest tests
====================================================== test session starts ======================================================platform linux -- Python 3.10.8, pytest-7.1.2, pluggy-1.0.0
Matplotlib: 3.6.0
Freetype: 2.6.1
rootdir: /home/neil/work/projects/topostats/TopoStats, configfile: pyproject.toml
plugins: mpl-0.16.1, typeguard-2.13.3, regtest-1.5.0, cov-3.0.0, forked-1.4.0, xdist-2.5.0
collected 2 items                                                                                                               

tests/test_a_theory.py FF                                                                                                 [100%]

=========================================================== FAILURES ============================================================_____________________________________________________________ test1 _____________________________________________________________
return_generator = array([0.52138574, 0.60384185, 0.4709418 , 0.20324794, 0.52875903,
       0.19103628, 0.2815456 , 0.75368155, 0.55167178, 0.86372208])
return_array = array([0.80537222, 0.24837266, 0.18985741, 0.98399558, 0.66999717,
       0.28038283, 0.20391323, 0.62506469, 0.65260432, 0.89880753])

    def test1(return_generator, return_array):
        print(f"generator   : {return_generator}")
        print(f"array       : {return_array}")
>       assert False
E       assert False

tests/test_a_theory.py:4: AssertionError
----------------------------------------------------- Captured stdout call ------------------------------------------------------generator   : [0.52138574 0.60384185 0.4709418  0.20324794 0.52875903 0.19103628
 0.2815456  0.75368155 0.55167178 0.86372208]
array       : [0.80537222 0.24837266 0.18985741 0.98399558 0.66999717 0.28038283
 0.20391323 0.62506469 0.65260432 0.89880753]
_____________________________________________________________ test2 _____________________________________________________________
return_generator = array([0.97476378, 0.15393237, 0.69908928, 0.44724145, 0.01751321,
       0.29102491, 0.38123661, 0.32102791, 0.94254467, 0.70266697])
return_array = array([0.13645032, 0.34320907, 0.8119946 , 0.148494  , 0.05932569,
       0.31441663, 0.42015645, 0.80801771, 0.00950759, 0.45408379])

    def test2(return_generator, return_array):
        print(f"generator   : {return_generator}")
        print(f"array       : {return_array}")
>       assert False
E       assert False

tests/test_a_theory.py:10: AssertionError
----------------------------------------------------- Captured stdout call ------------------------------------------------------generator   : [0.97476378 0.15393237 0.69908928 0.44724145 0.01751321 0.29102491
 0.38123661 0.32102791 0.94254467 0.70266697]
array       : [0.13645032 0.34320907 0.8119946  0.148494   0.05932569 0.31441663
 0.42015645 0.80801771 0.00950759 0.45408379]

---------- coverage: platform linux, python 3.10.8-final-0 -----------
Name                     Stmts   Miss  Cover
--------------------------------------------
tests/conftest.py           11      0   100%
tests/test_a_theory.py       8      0   100%
--------------------------------------------
TOTAL                       19      0   100%

==================================================== short test summary info ====================================================FAILED tests/test_a_theory.py::test1 - assert False
FAILED tests/test_a_theory.py::test2 - assert False
======================================================= 2 failed in 0.25s =======================================================
```

Not what I expected, as I expected the `array` printed by both `test1` and `test2` to be the same but the `generator` to be different. :thinking: 

Perhaps [`pytest-rng`](https://pypi.org/project/pytest-rng/) might be of use here rather than using `np.random.default_rng()` directly. :thinking:

MaxGamill-Sheffield, 2022-11-17:
Maybe related, but I see this issue primarily with the `pytest mpl` image tests.

Here there is a difference between running `pytest`, which on the main branch all pass, and `pytest <specified_test>` which can throw some errors. i.e. plot difference errors in `tests/test_grains_minicricle.py::test_cropped_image` complaining about differences in axes and titles for the "cropped image" test.

This occurs on:
Python 3.10.6
pytest-7.1.3
pluggy-1.0.0
Matplotlib: 3.6.2, 3.6.0, 3.5.3
Freetype: 2.6.1
plugins: 
 regtest-1.5.0 
 mpl-0.16.1, 0.16.0, 0.15.3
 cov-4.0.0

Additionally, when trying to generate new images using `pytest --lf --mpl-generate-path=tests/resources/img`, the tests still fail. Also, re-making the mpl images does nothing to resolve errors weirdly.

On the JPK branch, a test fails with `pytest` multiple fail with `pytest tests/test_grains_minicircle.py` and tests pass with `pytest --lf`.

ns-rse, 2022-12-23:
I believe this is down to how many times the Random Number Generator from Numpy is called.

I've made a [test repository](https://github.com/ns-rse/test-conftest) with code to demonstrate that the numbers generated differ based on the number of times the instantiated Random Number Generator has been called.

Why would this result in the observed behaviour described in this issue? Because if you are developing tests and running them in isolation then the arrays generated from the Random Number Generation will be perhaps only the first or second call to the Random Number Generation. However, when the whole test suite is then run the newly developed tests are unlikely to be the first ones called and the Random Number Generator will have already been called several times. This means the new tests will get a different set of numbers.  The same is true when using the `--lf` (last failed) option to re-run tests.

This is entirely expected behaviour with regards to the [Numpy Random Number Generator](https://numpy.org/doc/stable/reference/random/generator.html).

There are two solutions...

1) When developing tests that use fixture from `conftest.py` that returns random arrays run the whole test suite to generate the arrays used in your tests. Particularly relevant when generating Matplotlib images.
2) Alternatively instantiate a new Random Number Generator object for each test being written, with a fixed seed, and use that instead. This will over time mean tests have duplicated code though each time a new Random Number Generator object is generated and used.

closed 2022-12-23 by ns-rse