AFM-SPM/TopoStats #353: Re-Add Zero-Average Background Function

MaxGamill-Sheffield, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
Before the flattening improvements, there was a function which zero-averaged the background however, this is no longer present.

Upon first glance, this is because it seems the behaviour of the median flattening is to do this (see height scale bars below):
[image]
[image]

However, after testing with additional images, it seems this is not consistent. While the overall effect of the flattening pipeline is to do this, it is not done at the same step (see below). This is because applying the tilt removal afterwards, may not centre the image around 0:
[image]
[image]
[image]
[image]

A user adding their own flattening functions / running their own flattening pipeline may require this function, and/or this function may be required within this flattening pipeline for consistency between results.

**Describe the solution you'd like**
Re-add the average background:

- function:
```    
def average_background(self, image: np.ndarray, mask: np.ndarray = None) -> np.ndarray:
        """Zero the background

        Parameters
        ----------
        image: np.array
            Numpy array representing image.
        mask: np.array
            Mask of the array, should have the same dimensions as image.

        Returns
        -------
        np.ndarray
            Numpy array of image zero averaged.
        """
        medians = self.row_col_medians(image, mask)
        LOGGER.info(f"[{self.filename}] : Zero averaging background")
        return (image.T - np.array(medians["rows"], ndmin=1)).T
```
- Conftest
```
@pytest.fixture
def minicircle_zero_average_background(minicircle_masked_tilt_removal: Filters) -> Filters:
    """Zero average background"""
    minicircle_masked_tilt_removal.images[
        "zero_averaged_background"
    ] = minicircle_masked_tilt_removal.average_background(
        minicircle_masked_tilt_removal.images["masked_tilt_removal"], mask=minicircle_masked_tilt_removal.images["mask"]
    )
    return minicircle_masked_tilt_removal
```
```
@pytest.fixture
def minicircle_grain_gaussian_filter(minicircle_zero_average_background: Filters) -> Filters:
    """Apply Gaussian filter."""
    minicircle_zero_average_background.images["gaussian_filtered"] = minicircle_zero_average_background.gaussian_filter(
        image=minicircle_zero_average_background.images["zero_averaged_background"]
    )
    return minicircle_zero_average_background
```
```
- (and in `random_filters`)
```
- Filter Tests:
```
- (add one as there wasn't one previously)
```
- Minicircle Tests:
```
@pytest.mark.mpl_image_compare(baseline_dir="resources/img/")
def test_average_background(
    minicircle_zero_average_background: Filters, plotting_config: dict, plot_dict: dict, tmp_path
) -> None:
    """Test zero-averaging of background."""
    assert isinstance(minicircle_zero_average_background.images["zero_averaged_background"], np.ndarray)
    assert minicircle_zero_average_background.images["zero_averaged_background"].shape == (1024, 1024)
    assert minicircle_zero_average_background.images["zero_averaged_background"].sum() == 169375.4175476962
    plotting_config = {**plotting_config, **plot_dict["zero_averaged_background"]}
    fig, _ = Images(
        data=minicircle_zero_average_background.images["zero_averaged_background"],
        output_dir=tmp_path,
        pixel_to_nm_scaling=minicircle_zero_average_background.pixel_to_nm_scaling,
        **plotting_config,
    ).plot_and_save()
    return fig
```
**Describe alternatives you've considered**
Optionally, document the median row alignment should be done after tilt removal steps as this also has the additional functionality of aligning the row medians around 0.

SylviaWhittle, 2022-11-15:
Ah! Thank you, I had not considered this.

ns-rse, 2022-11-15:
Close by #360

PR #360 (2022-11-15): Maxgamill sheffield/353 zero background
Closes #353

This adds a function which gets the background mean height of the image (based on the mask) and substracts that mean from all data points so that the background is averaged around 0. 

A test has been added in test_filters.py but no tests added to the minicircle tests (filters and plotting_funcs) as @SylviaWhittle mentioned these will be replaced at some point. If I need to add them just let me know :)
closed 2022-11-15 by ns-rse